RedPitaya/RedPitaya #199: Streaming Server massive "Missed data buffers when reading from ADC"

tandersn, 2020-09-05:
I am trying the streaming server app and saving data. There seems to be a lot of missed data buffers, *more* when in 8 bit mode.  It is to the point where it appears that it misses so many samples, the sample rate is affected. When i view the file in gnu radio companion, i have to set a 31.25Mhz file to 25Mhz to see the frequencies i am capturing appear in the right place (i have a fixed signal at 7.1Mhz i am looking at).

b-swat, 2020-09-11:
May be. In the log file, you can see the number of lost buffers.
In the future, there are plans to add zero values to the output when data loss occurs.

tandersn, 2020-09-25:
> May be. In the log file, you can see the number of lost buffers.
> In the future, there are plans to add zero values to the output when data loss occurs.

That's one of the problems, the log file it says "zero" lost samples:

tandersn, 2020-09-25:
> May be. In the log file, you can see the number of lost buffers.
> In the future, there are plans to add zero values to the output when data loss occurs.

That's one of the problems, the log file it says "zero" lost samples: 
======================================================================
====   Data transfer report  2020-09-08 19:44:42 =====================
======================================================================
16u 31.25

Current decimation factor:	4

Missed data buffers when reading from ADC:	0

tandersn, 2020-09-25:
That's one of the problems, the log file it says "zero" lost samples: 

======================================================================
====   Data transfer report  2020-09-08 19:44:42 =====================
======================================================================
16u 31.25

Current decimation factor:	4

Missed data buffers when reading from ADC:	0

closed 2023-07-12 by b-swat