RedPitaya/RedPitaya #201: PWM not working from SCPI?

gsteele13, 2020-10-02:
Hi all,

It seems to work fine from the built-in Oscilloscope software, but if I specify `SOUR1:FUNC PWM`, then the output seems to do nothing? 

I ran this code with Out1 => In1 and Out2 => In2 with SMA cables (and 50 ohm terminator tee):

```
# Turn on Generator

wave_form = 'pwm'
freq = 1e4
ampl = 1

rp_s.tx_txt('GEN:RST')

for n in 1,2:
    rp_s.tx_txt(f'SOUR{n}:FUNC ' + str(wave_form).upper())
    rp_s.tx_txt(f'SOUR{n}:FREQ:FIX ' + str(freq))
    rp_s.tx_txt(f'SOUR{n}:VOLT ' + str(ampl))
    rp_s.tx_txt(f'OUTPUT{n}:STATE ON')
    
# phase is in radians!
phase = 0
rp_s.tx_txt(f'SOUR1:PHAS {phase/180*np.pi}')
rp_s.tx_txt('SOUR2:FUNC SINE')
rp_s.tx_txt('SOUR2:DCYL 0.2')

# Acquire signal

rp_s.tx_txt('ACQ:START')
rp_s.tx_txt('ACQ:TRIG NOW')

while 1:
    rp_s.tx_txt('ACQ:TRIG:STAT?')
    if rp_s.rx_txt() == 'TD':
        break

ch = []
for n in 1,2:
    rp_s.tx_txt(f'ACQ:SOUR{n}:DATA?')
    buff_string = rp_s.rx_txt()
    buff_string = buff_string.strip('{}\n\r').replace("  ", "").split(',')
    ch.append(list(map(float, buff_string)))

plt.plot(ch[0])
plt.plot(ch[1])
```

and this is the output:

[image]

Should I be digging into the source code of the SCPI server...?

Thanks,
Gary

gsteele13, 2020-10-02:
(also, it seems that the documentation page incorrectly specifies that the phase should be in degrees...?)

gsteele13, 2020-10-02:
this is my version info: 

Red Pitaya OS 0.98-617 / STEM 10

b-swat, 2020-12-09:
The documentation is correct, we update it according to our latest changes. Thus, please upgrade your board to the latest version.

closed 2022-07-17 by b-swat