RedPitaya/RedPitaya #205: Ethernet LED blinking changes DAC voltage

lneuhaus, 2021-01-13:
When the ethernet LED blinks, this results in a change in DAC voltage of the order of ~0.5 mV. We have confirmed this correlation with a photodetector placed near the ethernet port LED. I believe the linux OS ethernet driver is responsible for this blinking. Is there a way to disable it to get the best possible noise performance out of a redpitaya?

jeras, 2021-01-13:
Ethernet port blinking probably can be controlled through the devicetree.

This is the device tree file to modify:
https://github.com/RedPitaya/RedPitaya/blob/master/fpga/dts/ethernet.dtsi

This describes the available led options:
https://github.com/RedPitaya/linux-xlnx/blob/branch-redpitaya-v2017.2/include/dt-bindings/phy/phy-leds.h

This is the Lantiq PHY linux driver source:
https://github.com/RedPitaya/linux-xlnx/blob/branch-redpitaya-v2017.2/drivers/net/phy/lantiq.c

If the change could not be applied at runtime, then the simplest approach would be to create an alternative `.dtb` file, and overwrite the original on the SD card FAT partition.

It has been a couple of years since I last worked with devicetree, so you will probably have to find somebody else to help with how to properly apply it.

lneuhaus, 2021-01-14:
Thanks for these hints. I was able to solve it using https://github.com/PieVo/mdio-tool in the following way:

```
ssh root@my-redpitaya-hostname
git clone https://github.com/PieVo/mdio-tool.git
cd mdio-tool
cmake .
make
./mdio-tool w eth0 0x1b 0x0000
```

To re-enable blinking, run 
```
./mdio-tool w eth0 0x1b 0x0f00
```

The registers addresses and lots of more info can be found at https://datasheet.octopart.com/PEF-7071-V-V1.5-Lantiq-datasheet-62314344.pdf, e.g. page 126. 

If anyone knows how I have to modify the .dtsi-file that would be great!

lneuhaus, 2021-01-14:
Something like 

```

 
/*
 * Lantiq ethernet PHY 
 */
&gem0 {
	phy-handle = <&ethernet_0_mdio>;
	ethernet_0_mdio: ethernet_0_mdio@1 {
		reg = <1>;
		reg = <0x1b 0x0000>;
	};
};
```
maybe?

lneuhaus, 2021-11-10:
Is this fixed by https://github.com/RedPitaya/RedPitaya/commit/fbbbde82f3e65ba48bdd00048001af0a128414c1? Is a usage example for the command-line tool to disable ethernet LEDs available?

crtvalentincic, 2021-11-10:
This feature is available in the latest beta release 1.04-11, usage example can be seen here: https://redpitaya.readthedocs.io/en/latest/appsFeatures/command_line_tools/com_line_tool.html#led-enable-disable-utility

closed 2021-11-10 by lneuhaus