RedPitaya/RedPitaya #212: SDRlab 122-16: How to control GPIOs?

Apollo3zehn, 2021-02-10:
I try to control SDRlab 122-16 GPIOs. With another STEMLAB 125-14 I am able to load the mercury bitstream and then control the GPIO Pins using the SYSFS interface (https://redpitaya.readthedocs.io/en/latest/developerGuide/gpio/gpio.html). But there is no mercury or classic bitstream on the stable SDRlab 122-16 image and using one the other bitstreams in /opt/redpitaya/fpga doesn't work either. What should I do?

b-swat, 2021-02-20:
Hi
You can use fpga 0.94 with SCPI or api.
https://redpitaya.readthedocs.io/en/latest/appsFeatures/examples/dig-exm1.html#code-c
This examples uses LEDs. But you can change the constants to:
     RP_DIO0_P, //! <DIO_P 0
     RP_DIO1_P, //! <DIO_P 1
     RP_DIO2_P, //! <DIO_P 2
     RP_DIO3_P, //! <DIO_P 3
     RP_DIO4_P, //! <DIO_P 4
     RP_DIO5_P, //! <DIO_P 5
     RP_DIO6_P, //! <DIO_P 6
     RP_DIO7_P, //! <DIO_P 7
     RP_DIO0_N, //! <DIO_N 0
     RP_DIO1_N, //! <DIO_N 1
     RP_DIO2_N, //! <DIO_N 2
     RP_DIO3_N, //! <DIO_N 3
     RP_DIO4_N, //! <DIO_N 4
     RP_DIO5_N, //! <DIO_N 5
     RP_DIO6_N, //! <DIO_N 6
     RP_DIO7_N //! <DIO_N 7
for api
or
https://redpitaya.readthedocs.io/en/latest/appsFeatures/remoteControl/remoteControl.html#list-of-supported-scpi-commands
  {{DIO0_P ... DIO7_P}, {DIO0_N ... DIO7_N}}
for SCPI

Apollo3zehn, 2021-02-20:
Thanks for the samples! I have followed the API example and found that in the end it uses memory map (mmap) to talk to the FPGA. I am using a C# web application running on the board and would like to avoid needing to compile additional C code or create my own driver to reenable support for the Linux GPIO interface. Since I can use the mmap function in C# directly, I think I will follow that path (I have done it earlier for other signals using a custom FPGA image, so I am sure it will work). 

Is the reason that the Linux GPIO system is not supported anymore because it is deprecated?

I have found this: https://github.com/RedPitaya/RedPitaya/issues/126#issuecomment-327382216 and this: https://redpitaya.readthedocs.io/en/latest/developerGuide/gpio/gpio.html

Both are saying that the sysfs interface is deprecated, but not stating explicitly the differences between SDRlab 122-16 and STEMlab 125-14.

closed 2023-07-06 by b-swat