RedPitaya/RedPitaya #214: Some important links are broken and have been for a while.

fossum, 2021-03-02:
All of these links exist in a fresh install on the root GUI page.

List of broken links:

- Red Pitaya Store -> http://store.redpitaya.com/
- Feedback -> Opens blank Chrome tab.
- Instructions -> http://wiki.redpitaya.com/

Version: Red Pitaya OS 1.04-8

fossum, 2021-03-03:
If someone can point me in the right direction, I'd love to update these and submit a PR. I can't seem to find the app code for any of the apps listed.

fossum, 2021-03-04:
Okay. Now I'm really confused. I found apps-tools/assets/desktop.js which seems to already have the fix, but I have the latest beta and it's not fixed. I've loaded my SD card, but a search for desktop.js or grep for store produces nothing.

dzalf, 2022-09-11:
Any update on the missing links?
I just opened an issue to address the lack of response on a whole bunch of requests and PRs

Lightsaver7, 2023-09-07:
Hello,

Red Pitaya has moved the website/reorganized it. You can find the latest documentation here:
- https://redpitaya.com/
- Documentation => Techincal Documentation (we moved the wiki to Readthedocs which builds from GitHub)

If you want to access old documentation, please use the web archive (Wayback machine or another similar tool):
- http://web.archive.org/
- Type in the desired website