RedPitaya/RedPitaya #216: SendValue() in CStringParameter not overloaded

wuuuuuud, 2021-03-31:
For string type, CheckMinMax() should not be called as it will always return ""(empty string). I find that SetValue() is overloaded to bypass the check function, while SendValue() is not. I wonder if this is intended or a bug?

blupgnup, 2024-06-27:
I find myself annoyed with the same situation and it is becoming a problem if you want to return a CStringParameter to the frontend.
Indeed, the Set() function is overloaded but it does not set m_NeedSend to true (while SendValue() does).
Hence, we get the choice of a function that will work but not call a parameter update to the frontend and one that does not work.

By the way (as a workaround) is there a way to trigger the update of one parameter only (like SendAllParams() does for all parameters)? I cannot figure out such function.

b-swat, 2024-06-28:
> By the way (as a workaround) is there a way to trigger the update of one parameter only (like SendAllParams() does for all parameters)? I cannot figure out such function.

This is not the case in the current implementation.
It is possible to add such functionality, but there are no plans yet.
The code for CStringParameter really needs some improvement.