RedPitaya/RedPitaya #217: Cannot connect to application marketplace

arianjad, 2021-05-10:
Hi,

I am currently connecting to the RP via direct ethernet connection. I am able to access the web interface. However I cannot access the application marketplace. I receive the following error: "RedPitaya is offline and cannot access the server! Please follow quick start instructions to put it online."

I have tested with both 1.04-7 OS and 0.99-48, same issue. I have tried with both DHCP and static IP address, same issue.

satfan52, 2021-09-20:
Same issues after replacing the SoC by a new one bought on Aliexpress. The repaired board works fine except that it can't connect to the Marketplace.

crtvalentincic, 2022-02-15:
Please make sure that your RP has access to the internet. You can try to ping our website from RP to confirm this.
All our services will only work with original Red Pitaya boards.

Lightsaver7, 2023-09-07:
Red Pitaya needs to have internet access in order to access the application marketplace. If Red Pitaya is directly connected to the laptop/computer and the computer has internet access that does not necessarily mean that Red Pitaya has internet access. To resolve this issue, please make sure that Red Pitaya is directly connected to a router with internet access.

Red Pitaya requires an internet connection in the following instances:
- updating the OS with the Software update tool
- accessing application marketplace

To install the applications without an internet connection: 
- Download the ecosystem code from this repository
- Establish SSH connection with Red Pitaya
- Copy the desired application from the downloaded repository to the Red Pitaya's /opt/redpitaya/www/apps/ folder on the Red Pitaya's Linux OS
- Move to the /opt/redpitaya/www/apps/<copied api name>
- run the make command