RedPitaya/RedPitaya #221: Impedance analyzer app - link to guide in docs is dead

imanyakin, 2021-09-03:
Link referencing to docs looks dead: 
https://github.com/RedPitaya/RedPitaya/blob/master/apps-free/impedance_analyzer/info/info.json

These ones:
http://goo.gl/2lnbeN
https://wiki.redpitaya.com/index.php?title=User_Manual#LCR_meter_.28contributed.29

Lightsaver7, 2023-09-07:
I believe this is supposed to be the page the link leads, but the pictures cannot be recovered by the web archive:
[(http://web.archive.org/web/20160618182958/http://wiki.redpitaya.com/index.php?title=Impedance_analyzer)]

Lightsaver7, 2024-08-19:
We added impedance analyzer to the official Red Pitaya OS, you can find it here: https://redpitaya.readthedocs.io/en/latest/appsFeatures/applications/impedance/impedance.html#impedance-analyzer
This issue can be closed.