RedPitaya/RedPitaya #227: Jupyter notebook and python not working on 250-12

wolfallein, 2022-02-04:
Hi,

The jupyter notebook is not available inside the 'Development' section on the 250-12 version 1.04-24. However jupyter server seems to be running:

````
root@rp-ffff4e:~# systemctl status jupyter
● jupyter.service - Jupyter notebook server
   Loaded: loaded (/etc/systemd/system/jupyter.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-09-10 10:34:22 UTC; 3h 26min ago
 Main PID: 1626 (jupyter-noteboo)
   CGroup: /system.slice/jupyter.service
           └─1626 /usr/bin/python3 /usr/local/bin/jupyter-notebook --no-browser --allow-root

Sep 10 10:34:22 rp-ffff4e systemd[1]: Started Jupyter notebook server.
Sep 10 10:34:31 rp-ffff4e jupyter[1626]: [W 10:34:31.429 NotebookApp] All authentication is disabled.  Anyone who can connect to this server will be able to run code.
Sep 10 10:34:31 rp-ffff4e jupyter[1626]: [I 10:34:31.538 NotebookApp] Serving notebooks from local directory: /home/jupyter
Sep 10 10:34:31 rp-ffff4e jupyter[1626]: [I 10:34:31.539 NotebookApp] 0 active kernels
Sep 10 10:34:31 rp-ffff4e jupyter[1626]: [I 10:34:31.539 NotebookApp] The Jupyter Notebook is running at: http://127.0.0.1:8888/jupyter/
Sep 10 10:34:31 rp-ffff4e jupyter[1626]: [I 10:34:31.540 NotebookApp] Use Control-C to stop this server and shut down all kernels (twice to skip confirmation).
````

When trying to load using python3 I have:

````
root@rp-ffff4e:~# python3
Python 3.5.2 (default, Nov 17 2016, 17:05:23)
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from redpitaya.overlay.mercury import mercury as overlay
>>> fpga = overlay()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jupyter/RedPitaya/redpitaya/overlay/mercury.py", line 21, in __init__
    super().__init__(overlay='mercury')
  File "/home/jupyter/RedPitaya/redpitaya/drv/overlay.py", line 21, in __init__
    raise IOError('Device tree overlay source {} does not exist.'.format(dtbo))
OSError: Device tree overlay source /opt/redpitaya/fpga/mercury/fpga.dtbo does not exist.
>>>
````

What should I do to get it to work, at least in the Python console? `SCPI` is not an option for me because I need to upload waveform as fast as possible to the AWG.

Thanks in advance.

wolfallein, 2022-02-08:
Hi,

Does anyone from RP actually check the issues posted here? If not how do I get some support?

crtvalentincic, 2022-02-10:
Dear @wolfallein,
we apologize for the late reply, most of our support goes directly through our support email, issues are reported on Github 
are included on a wish list and we try to include them into OS releases when possible. 
Jupyter notebook is unfortunately not yet supported on the 250-12 version, but it is on the list for future improvements. There is also a plan to support JupyterLab that is a nice step further in the Jupyter evolution.
For the fast upload of AWG, we would recommend running Python script on RP using SCPI through localhost as explained in our quick-start https://redpitaya.readthedocs.io/en/latest/appsFeatures/remoteControl/remoteControl.html or creating a small C code command-line tool.
Thank you.

b-swat, 2023-07-06:
There are no plans to migrate the mercury project to 250-12 yet, but there is a plan to integrate  API from 0.94 to Jupyter

closed 2023-07-06 by b-swat