RedPitaya/RedPitaya #230: Unable to build latest relase: missing rp_hw

gkaretka, 2022-04-29:
On fresh downloaded image RedPitaya 125-14.

Running `make i2c` or `make spi` result in this.

```root@rp-f05742:/home/221053/test/RedPitaya/Examples/C# make i2c
cc -g -std=gnu11 -Wall -Werror -I/opt/redpitaya/include  -L/opt/redpitaya/lib  i2c.c  -static -lrp -lrp-hw -lm -lstdc++ -lpthread -o i2c
/usr/bin/ld: cannot find -lrp-hw
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'i2c' failed
make: *** [i2c] Error 1
```

Even running `make digital_led_blink` does the same.
```
root@rp-f05742:/home/221053/test/RedPitaya/Examples/C# make digital_led_blink
cc -g -std=gnu11 -Wall -Werror -I/opt/redpitaya/include  -L/opt/redpitaya/lib  digital_led_blink.c  -static -lrp -lrp-hw -lm -lstdc++ -lpthread -o digital_led_blink
/usr/bin/ld: cannot find -lrp-hw
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'digital_led_blink' failed
make: *** [digital_led_blink] Error 1
```

b-swat, 2022-04-29:
> On fresh downloaded image RedPitaya 125-14.
> 
> Running `make i2c` or `make spi` result in this.
> 
> ```
> cc -g -std=gnu11 -Wall -Werror -I/opt/redpitaya/include  -L/opt/redpitaya/lib  i2c.c  -static -lrp -lrp-hw -lm -lstdc++ -lpthread -o i2c
> /usr/bin/ld: cannot find -lrp-hw
> collect2: error: ld returned 1 exit status
> <builtin>: recipe for target 'i2c' failed
> make: *** [i2c] Error 1
> ```
> 
> Even running `make digital_led_blink` does the same.
> 
> ```
> root@rp-f05742:/home/221053/test/RedPitaya/Examples/C# make digital_led_blink
> cc -g -std=gnu11 -Wall -Werror -I/opt/redpitaya/include  -L/opt/redpitaya/lib  digital_led_blink.c  -static -lrp -lrp-hw -lm -lstdc++ -lpthread -o digital_led_blink
> /usr/bin/ld: cannot find -lrp-hw
> collect2: error: ld returned 1 exit status
> <builtin>: recipe for target 'digital_led_blink' failed
> make: *** [digital_led_blink] Error 1
> ```

Can you provide a link from where the latest version of the ecosystem was downloaded?
And also display a list of libraries in the folder: /opt/redpitaya/lib

b-swat, 2022-04-29:
Also write the version of the current ecosystem. It doesn't seem to have been updated.

gkaretka, 2022-04-29:
```
root@rp-f05742:~# ls /opt/redpitaya/lib
librp2.a   librp.a         librpapp_lcr.so  librp.so
librp2.so  librpapp_lcr.a  librpapp.so
```

gkaretka, 2022-04-29:
This can be viewed in http://rp-xxxxxx.local/ `[Red Pitaya OS 1.04-9 / STEMlab 125-14](http://rp-f05742.local/updater/)`

gkaretka, 2022-04-29:
Updated to `[Red Pitaya OS 1.04-11 / STEMlab 125-14](http://rp-f05742.local/updater/)` still does not work.

gkaretka, 2022-04-29:
So latest stable build does not work. Updating to beta(`[Red Pitaya OS 1.04-18 / STEMlab 125-14](http://rp-xxxxxx.local/updater/)`) solved this issue but IMHO stable build is supposed to be stable :)

b-swat, 2022-04-29:
> So latest stable build does not work. Updating to beta(`[Red Pitaya OS 1.04-18 / STEMlab 125-14](http://rp-xxxxxx.local/updater/)`) solved this issue but IMHO stable build is supposed to be stable :)

Everything is changing. If you take the example code for the corresponding release, then it will work.
The code in the master branch on the github corresponds to the latest release.
Well, I'll think about it in the future.

closed 2022-04-29 by b-swat