RedPitaya/RedPitaya #233: Python 3.5 reached end-of-life

bleykauf, 2022-05-23:
Currently RedPitaya OS uses python 3.5 which reached end of life (https://endoflife.date/python) more than a year ago.

As a consequence, some libraries are no longer compatible with 3.5 and many useful features of newer versions are not usable.

Are there any plans to upgrade to a supported version of python 3?

b-swat, 2022-07-04:
@bleykauf   Yes, we are currently working on migration to a new version of Ubuntu, including Python

bleykauf, 2023-02-08:
@b-swat: Is there any update on this or any way to help with this?

b-swat, 2023-02-16:
@bleykauf 
I am working on a new version of the unified ecosystem for all boards.
Here are the links to the pre-release version.
There ubuntu 22.04, updated python and fpga moved to 2020.1 vivado

Here is the pre-release version
Ubuntu: 
https://downloads.redpitaya.com/downloads/LinuxOS/red_pitaya_OS-beta_2.00.img.zip
Ecosystem:
https://downloads.redpitaya.com/downloads/Unify/ecosystems/

b-swat, 2023-03-27:
The latest version is python 3.10.6
[OS 2.0](https://downloads.redpitaya.com/downloads/Unify/RedPitaya_OS_2.00-15_beta.img.zip)

closed 2023-03-27 by b-swat