RedPitaya/RedPitaya #240: Command line streaming application

nbawden, 2022-07-12:
The data streaming app (rpsa_client) doesn't seem to connect when the streaming server is started via the command line. It works fine when the streaming server is started in the web interface though.

The documentation for the command line streaming application also looks to be slightly out of date; there is no `-c` option and the configuration file `/root/.streaming_config` is used by default.

b-swat, 2022-07-17:
@nbawden  Thank you. Yes it will be necessary to update the documentation on the server.
And what is the problem, why can't it connect to the server?
The console server needs to be launched via script. It will load the fpga overlay.
What version of the ecosystem are you using?
In the latest version, I added streaming console clients to the ecosystem itself.

nbawden, 2022-07-18:
@b-swat, Thank you for the response. I launch the server using:
```
LD_LIBRARY_PATH=/opt/redpitaya/lib
export LD_LIBRARY_PATH
cat /opt/redpitaya/fpga/fpga_streaming.bit  > /dev/xdevcfg
/opt/redpitaya/sbin/mkoverlay.sh stream_app
streaming-server
```
which just returns the message `streaming-server started`. When I run the streaming app, I get the usual messages about the memory and then a `Disconnect;` message shortly after (There's no `Try connect IP_ADDRESS` message like there is when the server is launched via the web app).
I'm using version 1.04-18 of the ecosystem; I cloned the repository only a few days ago.

b-swat, 2022-07-18:
Try this ecosystem

https://downloads.redpitaya.com/downloads/STEMlab-125-1x/ecosystems/ecosystem-1.04-28-93661995d.zip
https://github.com/RedPitaya/RedPitaya/tree/v1.04-28_125-14
In this version, the streaming code has been heavily redesigned. It should work better and more stable.

nbawden, 2022-07-19:
That works, thanks! The conversion to csv seems to produce tab delimited data though, e.g.
```
1	85	29
2	36	5
3	72	69
4	80	78
5	31	13
...
```
Any idea why?

b-swat, 2023-03-27:
In the latest version, I changed the csv format from tab to a comma.

closed 2023-03-27 by b-swat