RedPitaya/RedPitaya #245: Is this project still maintained?

dzalf, 2022-09-11:
I recently started playing around with an STM 125-14 which I got as a gift from a colleague of mine and I can already tell that it is an amazing tool.

I have noticed not only the high volume of open issues but also the number of days and months that they have been left open without a proper response.

Personally, I have also come across multiple links that are totally broken and no alternatives are provided. This occurs mainly in the documentation pages that used to be allocated in _readthedocs_

I believe that the community is keen to help however, the maintainers/developers must tell us what help they need from us

Any updates?

THANKS

dzalf, 2022-09-14:
Broken links examples:

1. The response links from this question in the forum are broken: https://forum.redpitaya.com/viewtopic.php?f=8&t=1800
2. Market pace is not reachable in STEMlab 125-14 on Red Pitaya OS 1.04-18

Getting error: "Cannot connect to Bazaar server"

b-swat, 2022-09-23:
Thanks for the feedback.
I checked. The links in that answer are very old (5 years have passed). Since that time, the documentation has changed in structure and links have changed.
https://redpitaya.readthedocs.io/en/latest/quickStart/SDcard/SDcard.html

You can also download images directly from the server.
https://downloads.redpitaya.com/

> Market pace is not reachable in STEMlab 125-14 on Red Pitaya OS 1.04-18

I checked this version of the image. Everything opened fine for me. Looks like somewhere a problem with the network on your side.

[image]
[image]

dzalf, 2022-10-18:
Thank you for your prompt response. 

I solved this by formatting and resintalling the latest OS on my SD Card.

I am now having issues with the board not recognising a compatible WiFi card. 

However, I will be submitting a new issue regarding this.

BW

closed 2023-07-06 by b-swat