RedPitaya/RedPitaya #249: Compilation errors

mravenca, 2023-03-09:
[ 10%] Building C object CMakeFiles/rp-hw-obj.dir/src/i2c-helper.c.o
In file included from /root/RPrelease-2023.1/RedPitaya-Release_2023.1/rp-api/api-hw/src/i2c-helper.c:5:0:
/usr/include/linux/i2c.h:68:8: error: redefinition of ‘struct i2c_msg’
 struct i2c_msg {
        ^
In file included from /root/RPrelease-2023.1/RedPitaya-Release_2023.1/rp-api/api-hw/src/i2c-helper.c:4:0:
/usr/include/linux/i2c-dev.h:37:8: note: originally defined here
 struct i2c_msg {
        ^
In file included from /root/RPrelease-2023.1/RedPitaya-Release_2023.1/rp-api/api-hw/src/i2c-helper.c:5:0:
/usr/include/linux/i2c.h:129:7: error: redefinition of ‘union i2c_smbus_data’
 union i2c_smbus_data {
       ^
In file included from /root/RPrelease-2023.1/RedPitaya-Release_2023.1/rp-api/api-hw/src/i2c-helper.c:4:0:
/usr/include/linux/i2c-dev.h:89:7: note: originally defined here
 union i2c_smbus_data {
       ^
/root/RPrelease-2023.1/RedPitaya-Release_2023.1/rp-api/api-hw/src/i2c-helper.c:6:23: fatal error: i2c/smbus.h: No such file or directory
compilation terminated.

b-swat, 2023-03-13:
Update ubuntu to RP_OS_2.0 (22.04)
this code is not built on the old version

mravenca, 2023-03-14:
Thanks, it helped

mravenca, 2023-03-16:
b-swat's advice helped.