RedPitaya/RedPitaya #254: Version 2.00 not booting on STEM125-14

wkumari, 2023-04-25:
This is largely a duplicate of #250 - I was not able to reopen the issue, so making a new one. 

I was not able to get v2.00 to boot on my (older) STEM125.-14. It would hang with `Failed to load 'dts//devicetree.dtb' `

My env were:

root@rp-f017d1:~# fw_printenv
bootcmd=run $modeboot
bootdelay=3
baudrate=115200
ipaddr=10.10.70.102
serverip=10.10.70.101
prod_date=12/22/13
kernel_image=uImage
ramdisk_image=uramdisk.image.gz
devicetree_image=devicetree.dtb
bitstream_image=system.bit.bin
loadbit_addr=0x100000
kernel_size=0x500000
devicetree_size=0x20000
ramdisk_size=0x5E0000
fdt_high=0x20000000
initrd_high=0x20000000
sdboot=echo Copying Linux from SD to RAM... && mmcinfo && fatload mmc 0 0x3000000 ${kernel_image} && fatload mmc 0 0x2A00000 ${devicetree_image} && fatload mmc 0 0x2000000 ${ramdisk_image} && bootm 0x3000000 0x2000000 0x2A00000
ethaddr=00:26:32:F0:17:D1
nav_code=4662
hw_rev=1.1.2
serial=155000063

I wasn't sure what to make the hw_rev, so I looked at the script and chose STEM_125-14_v1.1 - the silkscreen on the back of the board says v1.1, the `printenv` says 1.1.2, but the advice in #250 suggested STEM_125-14_v1.0. I have no idea if this was the appropriate version to choose, but, hey, it boots, so... :-P

I'm sure I'd never manually set the hw_rev, and it seems likely that many people have the form of 'hw_rev=<some numbers>', so perhaps this could be handled in the boot script?

b-swat, 2023-04-26:
1.1.2 is the wrong version of the model.
and replacing another version will not work normally. Board profiles use this value from hw_rev.

Probably you are using the first versions of the boards, where the version was not set according to the standard.

b-swat, 2023-04-26:
Here is an instruction on how to replace hw_rev
https://github.com/RedPitaya/RedPitaya/issues/250#issuecomment-1482360381

b-swat, 2023-04-26:
> I'm sure I'd never manually set the hw_rev, and it seems likely that many people have the form of 'hw_rev=', so perhaps this could be handled in the boot script?

we will complete the documentation and add instructions for updating hw_rev

b-swat, 2023-04-26:
> I looked at the script and chose STEM_125-14_v1.1 - the silkscreen on the back of the board says v1.1, the `printenv` says 1.1.2

Version 1.0 and 1.1 have no differences in functionality, except for version 125-10 and 125-14.
The only difference is in the circuit design.

If you specify 1.0 or 1.1, the functionality will not be affected.

closed 2023-07-06 by b-swat