RedPitaya/RedPitaya #259: Logic Analyser restarts after few seconds, in V 2.0 every time

BornemannR, 2023-05-28:
The logic Analyser WebFrontend restarts a new connection a few seconds after starting a measuremnet in 1.04-28-93661995d and every time immedatly in V2.0xx.
Tested with two older RepitayaBoards STEM-LAB125-14 (one from the initally Kickstarter-Version, one ~ 1 year later). 
It is not a reboot of the LinuxSytstem (e.g. voltage instabillity of the power supply). Same behaviour with or without the extern logic analyser module.
a short kmesg report: 

[ 3749.681137] rprx amba_pl:rprx@2: release
[ 3749.700181] rprx amba_pl:rprx@2: remove
[ 3749.737817] alloc_contig_range: [1d200, 1d2f2) PFNs busy
[ 3749.814115] xilinx-vdma 80420000.dma: Xilinx AXI VDMA Engine Driver Probed!!
[ 3749.837665] rprx amba_pl:rprx@2: No segment_size value in device tree. Setting to 4194304
[ 3749.852812] rprx amba_pl:rprx@2: No segment_count value in device tree. Setting to 2
[ 3749.864336] rprx amba_pl:rprx@2: reserved dma: dacd1740 and 0x2000 KiB @0xa000000
[ 3750.386247] rprx amba_pl:rprx@2: open
[ 3750.386297] rprx amba_pl:rprx@2: ioctl cmd:16 arg8
[ 3750.386306] rprx amba_pl:rprx@2: ioctl segment cnt set to 0x8
[ 3750.386332] rprx amba_pl:rprx@2: ioctl cmd:15 arg262144
[ 3750.386340] rprx amba_pl:rprx@2: ioctl segment size set to 0x40000
[ 3750.386380] rprx amba_pl:rprx@2: ioctl cmd:1 arg0
[ 3750.386387] rprx amba_pl:rprx@2: ioctl terminate all rx
[ 3750.690601] rprx amba_pl:rprx@2: release
[ 3752.618368] rprx amba_pl:rprx@2: remove
[ 3752.666015] alloc_contig_range: [1d200, 1d2f2) PFNs busy
[ 3752.737951] xilinx-vdma 80420000.dma: Xilinx AXI VDMA Engine Driver Probed!!
[ 3752.751186] rprx amba_pl:rprx@2: No segment_size value in device tree. Setting to 4194304
[ 3752.757994] rprx amba_pl:rprx@2: No segment_count value in device tree. Setting to 2
[ 3752.767743] rprx amba_pl:rprx@2: reserved dma: dacd1840 and 0x2000 KiB @0xa000000
[ 3753.288905] rprx amba_pl:rprx@2: open
[ 3753.288956] rprx amba_pl:rprx@2: ioctl cmd:16 arg8
[ 3753.288965] rprx amba_pl:rprx@2: ioctl segment cnt set to 0x8
[ 3753.288992] rprx amba_pl:rprx@2: ioctl cmd:15 arg262144
[ 3753.289000] rprx amba_pl:rprx@2: ioctl segment size set to 0x40000
[ 3753.289039] rprx amba_pl:rprx@2: ioctl cmd:1 arg0
[ 3753.289046] rprx amba_pl:rprx@2: ioctl terminate all rx

the Problem with the SN is solved like in Post #250: https://github.com/RedPitaya/RedPitaya/issues/250

b-swat, 2023-06-20:
How can we replicate this problem?
Maybe some kind of video how it happens?

BornemannR, 2023-06-21:
to replicate: try to record a digital signal, like a SPI or I2C Signal.

My settings and what i found: Browser is Opera Version:99.0.4788.65

It works with an older Firmware-Version:  Red Pitaya GNU/Linux Ecosystem, Version: 1.04, Build: 9

newer Version makes every time after starting the measurement a reload of the webpage

I have tried several power supplys to enshure that there was not a voltage problem, and tried with 2 older RedPitayas.

Connection from my Laptop directly with an EthernetCable to the RedPitaya.

b-swat, 2023-07-06:
I can't replicate this problem myself.
Can you upgrade to this  [version](https://downloads.redpitaya.com/downloads/Unify/nightly_builds/) and make a bug report, and then send it to the mail: support@redpitaya.com

BornemannR, 2023-10-29:
Hallo. 
Sorry for the late response. Perhaps it is a browser problem. I use Opera in the latest version.  Now i have try MS Edge in the standard installation. Wit the Edge browser it works. With the latest stable image (2.23 beta) i can't start the oscilloscope app with opera, but without any problems with MS Edge.
Best
Rainer

hyperglitch, 2023-10-31:
I had this issue too. Fixed by disabling ad blocker - it was blocking the fastclick.min.js file

Lightsaver7, 2024-01-25:
We recommend using the Google Chrome browser as we regularly provide updates for it. The other Chromium browsers are also supported but do not receive updates as often (we are a small team and have limited resources, so we decided to focus on Google Chrome). You can also try clearing the cookies and disabling the adblockers, that might help.