RedPitaya/RedPitaya #264: Can not open the Oscilloscope

LRSkylark, 2023-10-06:
As the title suggests I can not open the Oscilloscope App any more. When I first booted up the Red Pitaya 125-14 I was able to open the App with no issues, however after about a day it stopped working. Once it broke when I try to open the app it enter into an infinite loop of showing the loading page, then a blank white page and then it refreshes back to the loading page. 
I have tried the following to fix this problem
1) Fresh Install
2) Replacing the Oscilloscope with the latest nightly build
3) Replacing all apps with the latest nightly build

Is this a known issue? Is there a work around to fix this issue, as this is starting to cause a number of probelms?

Nightly build link I used, I downloaded the 5-10-2023 build, 2.00-162
https://downloads.redpitaya.com/downloads/Unify/nightly_builds/

b-swat, 2023-10-06:
Hi. Disable the AD block. It blocks the script and that's why nothing loads. I will fix it in the next versions.

LRSkylark, 2023-10-06:
> Hi. Disable the AD block. It blocks the script and that's why nothing loads. I will fix it in the next versions.

Thank you that was the problem.

closed 2023-10-06 by LRSkylark