RedPitaya/RedPitaya #266: Issues with the Lock-In+PID Application and Red Pitaya Forums

LRSkylark, 2023-10-27:
Issue with the Lock-In+PID Application - 
We are able to load the application, but once the application has been loaded, we are unable to change any settings within it to actually configure it to be used. And it normally either throws the error "Error While Sending Data (E3)" shown below or immediately resets the value back to default, not letting us actually change or used the feature.
I have tried the following to try and resolve this problem, any more suggestions would be appricated

-> 2 Pitayas
-> The latest released version of Unified OS, this has been re made multiple times, so there has been many clean installs
-> The Lock-In+PID from the app store
-> The Lock-In+PID from the github, installed over SCP
-> The Lock-In+PID from the github, installed directly onto the SD card using file explorer
-> Connecting to the device though it's IP Address
-> Connecting to the device through is "rp-mac address.local"  address

[image]

Forums Issue - 
I was going to ask this on the red pitaya forums rather than here, but I have been unable to activate my account on there for over 2 days now as the activation email is not sending, I have used 2 different email accounts.

b-swat, 2023-10-27:
What version of the ecosystem?

LRSkylark, 2023-10-27:
I am running the latest stable version of Unified OS, with the most up to date version off git hub or app store for the Lock-In+PID Application.

b-swat, 2023-10-27:
There is no support for this app in the latest version of the ecosystem.
At least fpga are not compatible in format bin and bit files. 
Ask the developer of this application to adapt to the new ecosystem.

LRSkylark, 2023-10-27:
Do you know what version of the ecosystem it does support? And if so, is there a link for us to download the version?

b-swat, 2023-10-27:
try this 
https://downloads.redpitaya.com/downloads/STEMlab-125-1x/

LRSkylark, 2023-10-27:
That seams to have fixed the problem thank you

closed 2023-10-27 by LRSkylark