RedPitaya/RedPitaya #267: Acquire fixed number of samples samples

KERRmie, 2023-11-04:
Hello,

I have a script working which uses a threshold to trigger data acquisition on the red pitaya. I want to adapt this to instead 'listen' for a fixed predefined number of samples? I have tried doing this using an acquisition duration variable, but I think the time taken for the code to run means that it is recording significantly more samples than I want.

Any suggestions are much appreciated!

Thanks.

b-swat, 2023-11-05:
Can you write part of script?
How many more samples are written to the buffer?

KERRmie, 2023-11-05:
Hi, 

Sorry I'm not too familiar with posting on GitHub, so I am hoping the code displays ok. It seems all that happens is the buffer is filled and then it exits the while loop and completes the remaining script. When I look at the saved file there are 16384 data points that have been captured.

`
# Define the Red Pitaya device IP address
rp_ip = 'rp-f0632f.local'

# Establish a connection to the Red Pitaya device
rp_s1 = scpi.scpi(rp_ip)

# Define the file path for saving data
filepath = 'C:\\Users\\jamie\\Documents\\WaveData'  # Change the path as needed

# Initialize data storage lists
DataFull2 = []

# Reset acquisition settings
rp_s1.tx_txt('ACQ:RST')

# Set acquisition parameters
no_samples = 5000
sampleRate = 5e6  # Set accordingly
decimationRate = 125e6 / sampleRate  # Calculate decimation rate for the defined sample rate
acquisition_duration = no_samples / sampleRate # calcs the duration of aqusiton for the given no samples and sample rate
trigThresh = 5e-5 # trigger threshold

# Send acquisition parameters to Red Pitaya
rp_s1.tx_txt('ACQ:DATA:FORMAT ASCII')
rp_s1.tx_txt('ACQ:DATA:UNITS VOLTS')
rp_s1.tx_txt(f'ACQ:DEC {decimationRate}')
rp_s1.tx_txt(f'ACQ:TRIG:LEV {trigThresh}')

# Define trigger conditions
rp_s1.tx_txt('ACQ:TRIG CH2_PE')

# Datetime calculations
current_time = datetime.datetime.now()  # Get the current time
print("While loop starts:", current_time)

# Main acquisition loop
while True:
    # Start acquisition
    rp_s1.tx_txt('ACQ:START')
    acq_start_time = time.time()  # Record acqusition start time 
    print(acq_start_time)
    while True:
        rp_s1.tx_txt('ACQ:TRIG:STAT?')
        if rp_s1.rx_txt() == 'TD':
            break

    # Acquiring data for Channel 22
    rp_s1.tx_txt('ACQ:SOUR2:DATA?')
    buff_string2 = rp_s1.rx_txt()
    buff_string2 = buff_string2.strip('{}\n\r').replace("  ", "").split(',')
    buff2 = list(map(float, buff_string2))
    DataFull2 += buff2  # Channel 2

    # Calculate the time elapsed since the trigger event
    elapsed_time = time.time() - acq_start_time
    
    # Check if the acquisition duration is reached
    if elapsed_time >= acquisition_duration:
        break

# Check time the while loop completes
completion_time =  datetime.datetime.now() 
print("While loop ends:", completion_time)

# Plot the acquired data (optional)
import matplotlib.pyplot as plt
plt.plot(DataFull2)
plt.title('Sensor_Ch2')
plt.show()

# Saving Data
df = pd.DataFrame()
df['RP1_Ch2'] = DataFull2
df.to_csv(filepath + '\\RP_PLB3.csv', index=False) 

 `

b-swat, 2023-11-05:
As I understand it, first you need to look at the list of SCPI commands and acceptable parameters.

https://redpitaya.readthedocs.io/en/latest/appsFeatures/remoteControl/remoteControl.html#list-of-supported-scpi-commands

_decimationRate = 125e6 / sampleRate # Calculate decimation rate for the defined sample rate_

such decimation is not supported.

KERRmie, 2023-11-05:
Thanks for pointing that out. I've had a read of the documentation, but it I cant see a command which will allow me to acquire a fixed number of samples.

b-swat, 2023-11-06:
ACQ:SOUR<n>:DATA:Old:N? <m>

b-swat, 2023-11-06:
and it looks like you want to capture 1 sample per second, for this purpose it is better to use 
https://github.com/RedPitaya/RedPitaya/blob/master/Examples/python/analog_inputs.py
The high-speed input can acquire data from at least 1.9kHz

closed 2023-11-07 by KERRmie