RedPitaya/RedPitaya #268: stuck at "Getting list of applications" screen and restarts

Hermy07, 2023-11-08:
I just tried out the last versions, Stable version (2.00-18) and Beta version (2.00-23), and stuck at the Getting list of applications screen. 
Also it seems the board resets if i am trying to connect. 
Using board 125-14 v1.0

b-swat, 2023-11-08:
Disable ad block

Hermy07, 2023-11-09:
unfortunately it is not the ad block and it is working with older versions like 1.4

b-swat, 2023-11-09:
can you check fw version in eeprom?

b-swat, 2023-11-09:
https://github.com/RedPitaya/RedPitaya/issues/254

Hermy07, 2023-11-10:
Is there a preferred way to do this? 
Also checked the environment with printenv, but there is no hw version listed

Hermy07, 2023-11-10:
sorry wrong command
fw_printenv: hw_rev=STEM_125-14_v1.0

b-swat, 2023-11-10:
Try to connect the board via console/minicom. you can see if the board reboots when connected

Hermy07, 2023-11-10:
[image]

The output just after bootup and after i open the webserver in browser

b-swat, 2023-11-13:
I've only observed two cases in all my time. The reboot can be caused by a frozen FPGA. When the FPGA is not from another board to write for example from Z7020 or to write when the FPGA registers are being accessed. The second case is when there is insufficient power supply. If the power supply is bad, then increasing the load on the processor leads to the fact that the board immediately reboots.

arnaudrichard, 2023-11-13:
I confirm that powering RedPitaya by a PC USB or even worse through USB hub lead to spurious resets.

Lightsaver7, 2023-11-17:
Please use the power supply you received together with the board or use a 5V DC power supply that can provide 2 Amps of current (for completely independent operation (couple of hours) you can also use phone power banks). The USB from the computer supplies the necessary voltage, but the current is too low (approx. 1.2 A on USB 3.0).

Hermy07, 2023-12-06:
A insufficient power supply was something i thought about. I tried it with power supplies with much higher output power.
It seems like the board i used is damaged. I tried it with other v1.0 boards and it worked just fine.

closed 2024-01-11 by b-swat