RedPitaya/RedPitaya #269: RedPitaya OS 2.0 not booting on 125-14 #250

warp-bit, 2023-11-25:
Hi b-swat,

not sure , if it would be more appropriate to just comment on the existing thread (RedPitaya OS 2.0 not booting on 125-14 #250 ) or opening this new one, it's my first contact to this community.

I have the same issue, but neither fix (setting hw_rev=STEM_125-14_v1.0 or setting hw_rev=STEM_125-14_v1.1) resolved it.

My output fw_printenv is:

root@rp-f08c4f:~# fw_printenv 
bootcmd=run $modeboot
bootdelay=3
baudrate=115200
ipaddr=10.10.70.102
serverip=10.10.70.101
prod_date=12/22/13
kernel_image=uImage
ramdisk_image=uramdisk.image.gz
devicetree_image=devicetree.dtb
bitstream_image=system.bit.bin
loadbit_addr=0x100000
kernel_size=0x500000
devicetree_size=0x20000
ramdisk_size=0x5E0000
fdt_high=0x20000000
initrd_high=0x20000000
sdboot=echo Copying Linux from SD to RAM... && mmcinfo && fatload mmc 0 0x3000000 ${kernel_image} && fatload mmc 0 0x2A00000 ${devicetree_image} && fatload mmc 0 0x2000000 ${ramdisk_image} && bootm 0x3000000 0x2000000 0x2A00000
ethaddr=00:26:32:F0:8C:4F
nav_code=4702
serial=162400000
hw_rev=STEM_125-14_v1.1
root@rp-f08c4f:~# fw_setenv hw_rev STEM_125-14_v1.0
root@rp-f08c4f:~# fw_printenv 
bootcmd=run $modeboot
bootdelay=3
baudrate=115200
ipaddr=10.10.70.102
serverip=10.10.70.101
prod_date=12/22/13
kernel_image=uImage
ramdisk_image=uramdisk.image.gz
devicetree_image=devicetree.dtb
bitstream_image=system.bit.bin
loadbit_addr=0x100000
kernel_size=0x500000
devicetree_size=0x20000
ramdisk_size=0x5E0000
fdt_high=0x20000000
initrd_high=0x20000000
sdboot=echo Copying Linux from SD to RAM... && mmcinfo && fatload mmc 0 0x3000000 ${kernel_image} && fatload mmc 0 0x2A00000 ${devicetree_image} && fatload mmc 0 0x2000000 ${ramdisk_image} && bootm 0x3000000 0x2000000 0x2A00000
ethaddr=00:26:32:F0:8C:4F
nav_code=4702
serial=162400000
hw_rev=STEM_125-14_v1.0

Any suggestions or help would be greatly appreciated!

Kind regards 
warp-bit

b-swat, 2023-11-26:
Can you attach the board loading log? You need to connect via the console:
https://redpitaya.readthedocs.io/en/latest/developerGuide/software/console/console/console.html

warp-bit, 2023-11-26:
Finally loaded the 2.0 image on a fresh new SD-Card and - voila - the board booted fine.

Strangely enough the other card seems to be only "broken a tlitte" - it is still possible to flash my former RedPitaya 4.18 image onto it and boot it succesfully, too.

So thanks for your immediate reply and the hint on using the console!

closed 2023-12-19 by b-swat

b-swat, 2024-02-08:
@warp-bit  
I found a critical bug that prevented the board from booting correctly.
Corrected version 223 and higher
https://downloads.redpitaya.com/downloads/Unify/nightly_builds/