RedPitaya/RedPitaya #272: How to set trigger mode in Python?

karu2003, 2024-01-11:
How to set trigger mode in Python?

This example doesn't work
https://redpitaya.readthedocs.io/en/latest/appsFeatures/examples/acquisition_generation/acq_genRF-exm1.html

I need to set it Trigger to single mode. in WEB GUI. :(

b-swat, 2024-01-12:
Hello. It is not clear what task is meant.
Trigger mode parameters can be found here.
https://redpitaya.readthedocs.io/en/latest/appsFeatures/remoteControl/command_list.html#acquisition-trigger

karu2003, 2024-01-12:
> Hello. It is not clear what task is meant. Trigger mode parameters can be found here. https://redpitaya.readthedocs.io/en/latest/appsFeatures/remoteControl/command_list.html#acquisition-trigger

I haven’t found how to control AUTO, Normal, single. :(  how to set r/f edge?

b-swat, 2024-01-12:
"AUTO, Normal, single" there is no such functionality in the API. SCPI commands for basic interaction with FPGA.

Here is a simple example of signal capture on 1 channel in rising edge mode. This example is similar to the behavior of SIngle mode
https://github.com/RedPitaya/RedPitaya/blob/master/Examples/python/acquire_trigger_posedge.py

karu2003, 2024-01-12:
> "AUTO, Normal, single" there is no such functionality in the API. SCPI commands for basic interaction with FPGA.
>
> Here is a simple example of signal capture on 1 channel in rising edge mode. This [...]

Thanks, I found this myself and tried it. This is not enough for me. I need full remote control. Is it possible to change parameters via ssh.

closed 2024-01-12 by b-swat