RedPitaya/RedPitaya #273: How to get repeatable results?

karu2003, 2024-01-12:
I have a problem with RedP getting the same result.
Example of my code. 
https://github.com/karu2003/redp_examples/blob/main/acq_diff.py

I try to read the signal a hundred times. Using an external 500KHz generator and hoping to get the same diagram. But this doesn't happen. The trigger works anywhere, rarely at the same time. I need a large number of diagrams to create an AI model.

What could be the problem?

[image]

b-swat, 2024-01-12:
This wrong:

```
rp_s.tx_txt('ACQ:START')
rp_s.tx_txt('ACQ:TRIG CH1_PE')

for i in range(100):
```

Must be:
```
for i in range(100):
  rp_s.tx_txt('ACQ:START')
  rp_s.tx_txt('ACQ:TRIG CH1_PE')
```
To receive a new portion of data, you need to reinstall the trigger.

As I understand it, the task is to obtain a large data buffer.
Check out this module:
https://redpitaya.readthedocs.io/en/latest/appsFeatures/applications/streaming/appStreaming.html

karu2003, 2024-01-13:
Thanks, this doesn't solve the problem.
I need to close the WEB GUI.
How can i stop the WEB GUI via  API?
So that data is sent only to one source.

Lightsaver7, 2024-01-15:
Please do not use the SCPI server together with the Oscilloscope (or any other application). This can result in unidentified behaviour. The best way is to just start the SCPI server and leave it open.

I just noticed that we documented the _OLD:N (ACQ:SOUR1:DATA:Old:N)_ and _LAT:N (ACQ:SOUR1:DATA:Last:N)_ wrong. This will be fixed. OLD:N reads data form the start of the buffer onwards (oldest N samples) and LAT:N reads data from the end of the buffer (latest N samples)

As for fixing your code:
- Set the trigger delay to 8192 (put the _ACQ:START_ and _ACQ:TRIG_ commands inside for loop) and use the _ACQ:SOUR1:DATA:OLD:N? <data_length>_ command
- Set the trigger delay to _-(8192-<data length>)_ (put the _ACQ:START_ and _ACQ:TRIG_ commands inside for loop) and use the _ACQ:SOUR1:DATA:LAT:N? <data_length>_ command (where data length)

Either of these two should work.

closed 2024-05-23 by b-swat