RedPitaya/RedPitaya #274: documentation of the SPI interface.

karu2003, 2024-02-29:
Question about documentation and use of the SPI interface.
I plan to use RedPitaya as a tester. I need to check the external DAC and ADC. (DAC70501, ADS7056).
RedPitaya works like a generator and GPIO.
what is the maximum SPI speed? I need at  ADS7056 least 1 MHz sample rate - that's up to 50 MHz clock on SPI.
Is 18..20..24 bit frame width possible?
what is the maximum message size? 1K...16K or what?
I use SCPI.

Thank you.
Andrew.

Lightsaver7, 2024-02-29:
Hello Andrew,

You can find more info on SPI SCPI commands here:
https://redpitaya.readthedocs.io/en/latest/appsFeatures/remoteControl/command_list.html#spi

The maximum should be 100 MHz clock speed (I need to double-check with the team).
You can transfer a maximum of 8 bits per package (SPI word length can be either 7 or 8 bits), so if you have a 24-bit frame it will have to be split and sent in three packets (as 3 bytes), then merged again on the other side (this is how SPI communication works).
I think the length of the SPI message is up to you. As long as you define the message length correctly at the start then you can set it to almost anything. You can check the source code of the "rp_SPI_SetBufferForMessage" function on the link below to see whether there are any limitations.

The SCPI commands work through the C API commands (which are executed in the background). You can check them out here:
https://github.com/RedPitaya/RedPitaya/blob/master/rp-api/api-hw/include/rp_hw.h
You can find the source code in the vicinity of the file above.

Good luck with your project!
Miha

karu2003, 2024-02-29:
"(SPI word length can be either 7 or 8 bits)"
this means that I cannot use the CS signal and must generate it myself. This can be done to control the DAC.
But managing the CS to read the ADC is not at all suitable.
I need to check how the commands work.  
SPI:MSG<n>:TX<m>:RX:CS <data>
SPI:MSG<n>:RX<m>:CS
If the CS occurs after the entire message, i can work with only one frame.
But what sample rate will be in this case if I need 1K...16K of such packages is not clear.
It will be very difficult to check the bandwidth.

karu2003, 2024-02-29:
I tried your example to record messages over SPI. The CS is valid for the entire message. This makes me happy.
But. The duration of this signal surprised me.   Can this be reduced somehow?
[image]

b-swat, 2024-03-01:
@karu2003  Hi
As I remember, the :CS suffix saves the CS signal level between messages.
I also think that after transmitting the last message the driver does not change the level.
And on the oscilloscope it changed the level when we closed the connection, I think so. This is a big delay.
You can try to force another blank message to be added to the end of the message without the :CS suffix. Which will force the driver to change the level.

Like so
SPI:MSG:CREATE 2
SPI:MSG1:TX0:CS 0,0
SPI:MSG1:RX0
SPI:PASS

closed 2024-07-15 by b-swat