RedPitaya/RedPitaya #275: sample rate or decimation.

karu2003, 2024-03-02:
The more I work with RedPitaya, the more doubts I have about my choice.
It is not possible to obtain Sweep via SCPI.
I'm trying to use 'arbitrary'. But. I'm confused about where and what the connection is in the generator with the sample rate and decimation.
What frequency is meant to create your Waveform? Is this sample rate or period?
There is no signal at the generator output.

Thanks if you can help me figure it out.
Andrew.

https://github.com/karu2003/pg_ui/blob/master/preamp_tester.py
https://github.com/karu2003/pg_ui/blob/master/redpctl.py

def arbitrary

b-swat, 2024-03-04:
Check this example 
https://github.com/RedPitaya/RedPitaya/blob/master/Examples/python/generate_arbitrary_waveform.py

The generator operation algorithm is as follows (using the example of sine):
- A data buffer of 16 * 1024 samples is created.
- This buffer is filled with a sine signal of 1 period and an amplitude of 1V
- The amplitude multiplier and calibration in the FPGA are set through registers
- Sets the signal generation step.
- Starting the generator.

Regarding the signal frequency.
The sample generator step is set using the following formula.
step =  round(65536 * frequency / 125e6 * (16 * 1024));

That is, the FPGA moves through the cyclic buffer with a certain step, and this step sets what frequency the signal will be.

PS
The buffer can be less than 16k samples, but it cannot be more.
The solution is not very high quality; in the future we will think about increasing the buffer size.

PS2
https://downloads.redpitaya.com/downloads/Unify/nightly_builds/
In the latest version I added commands for sweep mode in scpi.
https://redpitaya.readthedocs.io/en/latest/appsFeatures/remoteControl/command_list.html#sweep-mode-extended

This mode is similar to the mode in the oscilloscope web application.

Lightsaver7, 2024-03-05:
Hello Karu,

I suggest you check the SCPI examples here:
https://redpitaya.readthedocs.io/en/latest/appsFeatures/remoteControl/examples_top.html#examples

As for why the signal does not appear on the output, the most common issue is that the output signal is enabled (OUTPUT<n>:STATE ON), but the generation is not triggered (SOUR<n>:TRIG:INT).

The arbitrary waveform generator accepts ONE period of your custom signal (saved in 16384 sample long buffer (16*1024)), which is then generated with the specified frequency (SOUR<n>:FREQ:FIX), amplitude (SOUR<n>:VOLT), and other applicable settings. You can also use the predefined waveforms such as SINE, TRIANGLE, SQUARE, etc. through the SOUR<n>:FUNC command.

I hope this helps.

Lightsaver7, 2024-03-13:
@karu2003 In the latest Nightly Builds we added an option to control the SWEEP mode parameters with the SCPI commands. Check the log here and the readme instructions for how to install the latest Nightly build:
https://redpitaya.readthedocs.io/en/latest/quickStart/SDcard/SDcard.html#nightly-builds