RedPitaya/RedPitaya #276: resolution of the generator output?

karu2003, 2024-03-09:
How to correctly understand the resolution of the generator output?
14 Bit > 1V. 
If I need -20db, what resolution will the generator have?
14bit be or what?

b-swat, 2024-03-11:
14 bits is for 1 volt. The amplitude reduction is digital, that is, at low amplitudes the bit depth will be lower.

1V / 8192 (13 bit) = 0.00012207 V/bit
0.5V / 0.00012207 = 4096 (12 bit)

karu2003, 2024-03-12:
> 14 bits is for 1 volt. The amplitude reduction is digital, that is, at low amplitudes the bit depth will be lower.
> 
> 1V / 8192 (13 bit) = 0.00012207 V/bit Reducing the signal to 0.5 volts 0.5V / 0.00012207 = 4096 (12 bit)

Thank you. I didn't expect this method. I was hoping that you were changing the reference voltage to DAC or so. :( 
Then there would be 14 bits for any voltage. 
I need  to install an external attenuator.

closed 2024-05-23 by b-swat