RedPitaya/RedPitaya #279: several devices on the i2ts bus?

karu2003, 2024-03-12:
How to easily ensure reading and writing of several devices on the i2ts bus.
I connected MCP23008 and LTC1380.
With MCP I can only read once. LTC1380 is not detected yet.

https://github.com/karu2003/pg_ui/blob/master/MCP230XX.py

https://github.com/karu2003/pg_ui/blob/master/redpctl.py

    def read_byte_data(self, i2cAddress, reg):
        self.rp_s.tx_txt('I2C:DEV%d "/dev/i2c-0"' % i2cAddress)
        self.rp_s.tx_txt("I2C:FMODE ON")
        self.rp_s.tx_txt('I2C:Smbus:Read%d' % reg)
        value = int(self.rp_s.rx_txt())
        return value

    def write_byte_data(self, i2cAddress, reg, regValue):
        self.rp_s.tx_txt('I2C:DEV%d "/dev/i2c-0"' % i2cAddress)
        self.rp_s.tx_txt("I2C:FMODE ON")
        self.rp_s.tx_txt("I2C:Smbus:Write" + str(reg) +" "+ str(regValue))
        return

b-swat, 2024-03-13:
This command will help you understand whether there are devices connected to the bus or not.
i2cdetect -y -r 0
In this case, I can't help you.

b-swat, 2024-03-13:
Please note that some chips do not work on the SM bus. This is not direct reading and writing to registers, but its own protocol

karu2003, 2024-03-13:
"but its own protocol" --- How should this be understood? Users can't use it for themselves? I want to read 10 registers. With MCP.
Doesn't read reddit.
Nothing happens on the bus at all.

There are signals on the bus but it doesn't read anything.

What protocol do you support? i2ts or smbus?

karu2003, 2024-03-13:
I see two errors: the command I2C:Smbus:Read2 works without a question mark. :( 
I monitor everything with an i2c analyzer and see that the chip is sending data.
or i2cget -y 0 0x20 0x00. 
the command rp_s.rx_txt() does not provide any data. and works without timeout.
which causes the program to stop.

karu2003, 2024-03-13:
this example doesn't work.
https://github.com/RedPitaya/RedPitaya/blob/master/Examples/python/i2c_eeprom_read_calib_125_14.py

my firmware
##############################################################################
# Red Pitaya GNU/Linux Ecosystem
# Version: 2.00-43f2d6cd9
# Build: 23
# Branch: 
# Commit: 43f2d6cd994dfc684cd1e5e1c3ce85bab91e8e44
# U-Boot: "redpitaya-v2022.1"
# Linux Kernel: "branch-redpitaya-v2022.3"
# Pro Applications: ""
##############################################################################

karu2003, 2024-03-13:
the same doesn't work.                                                                                         
# Red Pitaya GNU/Linux Ecosystem                                                                                                                                             
# Version: 2.00-73ff4bc8f                                                                                                                                                    
# Build: 30                                                                                                                                                                  
# Branch:                                                                                                                                                                    
# Commit: 73ff4bc8f14b33a839410eaf29205c2b8c8cd7a9                                                                                                                           
# U-Boot: "redpitaya-v2022.1"                                                                                                                                                
# Linux Kernel: "branch-redpitaya-v2022.3"                                                                                                                                   
# Pro Applications: ""

b-swat, 2024-03-13:
this old example. 
We know about it.
There are new example
https://github.com/RedPitaya/RedPitaya/blob/master/Examples/C/i2c_eeprom_read_calib.c

Lightsaver7, 2024-03-13:
Hello @karu2003,

Thank you for reporting this.

We will update the SCPI example (and other digital communication tutorials) in the near future. You can also run Python code directly on the board itself, and there you can use the same functions as in C (check the example above). 

Please check the command list if a specific command is not working for you (the commands are up-to-date there):
https://redpitaya.readthedocs.io/en/latest/appsFeatures/remoteControl/command_list.html#list-of-supported-scpi-api-commands

Please also check this Jupyter example to familiarize yourself with the order of calibration parameters in 2.00 OS:
https://github.com/RedPitaya/jupyter/blob/redpitaya_OS_2.x_rp_api/examples/hardware/calibration.ipynb

karu2003, 2024-03-13:
> Hello @karu2003,
>
> Thank you for reporting this.
>
> We will update the SCPI example (and other digital communication tutorials) in the near future. You can also run Python code directly on the board [...]

I2C:Smbus:Read?
rp_s.rx_txt()
i use SCPI.
I already wrote that it doesn’t work.
What is the likelihood that you can fix it quickly?
Two days or a month?

karu2003, 2024-03-13:
in 1.04  work!!! without a question mark.
    def read_byte_data(self, i2cAddress, reg):
        self.rp_s.tx_txt('I2C:Smbus:Read%d' % reg)
        value = self.rp_s.rx_txt()
        value = int(value)
        return value

b-swat, 2024-03-14:
Now this is interesting. What version 1.04 does the command work in?

b-swat, 2024-03-14:
I tested 250-12 on my board.
There is a MAX7311 chip on the board itself at address 0x20

[image]

The  command "I2C:S:R2?" works correctly.

[image]

[image]

The fact that in version 1.04 there is no "?" in the request. I know about this, in version 2.00 I corrected these commands.

karu2003, 2024-03-14:
i have 125-14.

b-swat, 2024-03-14:
it doesn't matter since the code is the same for all boards

karu2003, 2024-03-14:
Just try only reading.
I take your example. The code doesn't work.

b-swat, 2024-03-14:
What are the errors?
self.rp_s.tx_txt_check_error('I2C:Smbus:Read%d' % reg)

karu2003, 2024-03-14:
'I2C:Smbus:Read%d?' % reg 
I don't see any signals on the bus.

b-swat, 2024-03-14:
The difference between 1.04 and 2.00 is in the Linux kernel version, maybe there is some kind of problem there.

b-swat, 2024-03-14:
You can attach a diagram for connecting the chip to the RP. I'll try to repeat this problem as soon as I get the chip.

karu2003, 2024-03-14:
I connect an oscilloscope to the bus. I see the initialization of REDPI. Then I run calibration reading and see nothing. I don't need to connect the MCP. I have two REDPIs, both behave the same.

Lightsaver7, 2024-03-14:
Hello @karu2003,

The "I2C:Smbus:Read%d" command changed between versions 1.04 and 2.00. In 1.04 the command does not have a question mark at the end, while in 2.00 there is a question mark at the end. This change was implemented to make the command more inline with other commands that request data.

As far as the oscilloscope not working, I assume you updated the OS from 1.04 to 2.00. Please check whether you performed the calibration reset to "factory default settings" as it says in the note here:
https://redpitaya.readthedocs.io/en/latest/quickStart/SDcard/SDcard.html#os

Please also check whether the input signals in the oscilloscope are not hidden in the settings (Go to IN1 (cog wheel) and make sure "shown" is highlighted).

b-swat, 2024-03-14:
Update to the latest, including Linux
https://downloads.redpitaya.com/downloads/Unify/RedPitaya_OS_2.00-35_beta.img.zip

karu2003, 2024-03-14:
Thank you. Works.
LTC1380 also works using the Buffer.  tx_txt("I2C:IO:W:B1 " + str(regValue))

The system doesn't find LTC1380.
But it works. :(  :)

closed 2024-03-14 by b-swat