RedPitaya/RedPitaya #284: Trigger and buffer trouble with SCPI in Python

pjcoria, 2024-04-11:
The idea is to acquire a signal using one of the input channels triggering with the same channel. The problem is that I can't get the trigger working properly, it appears to ignore it and the data plots have discontinuities as shown in the following pictures:

[image]

[image]

[image]

This is the code used to get and plot the data and is based on the code examples for SCPI found in the Red Pitaya Documentation:

`import sys
import redpitaya_scpi as scpi
import matplotlib.pyplot as plot
import numpy as np

IP = 'rp-f04df7.local'

rp_s = scpi.scpi(IP)

rp_s.tx_txt('ACQ:RST')
rp_s.tx_txt('ACQ:STOP')

rp_s.tx_txt('ACQ:DATA:FORMAT ASCII')
rp_s.tx_txt('ACQ:DATA:Units VOLTS')

rp_s.tx_txt('ACQ:DEC 1')
rp_s.tx_txt('ACQ:TRig CH1_PE')
rp_s.tx_txt('ACQ:TRig:LEV 0.5 V')
rp_s.tx_txt('ACQ:TRig:DLY 0')

rp_s.tx_txt('ACQ:START')
#rp_s.tx_txt('ACQ:TRig NOW')

while 1:
    rp_s.tx_txt('ACQ:TRIG:STAT?')
    if rp_s.rx_txt() == 'TD':
        break

while 1:
    rp_s.tx_txt('ACQ:TRIG:FILL?')
    if rp_s.rx_txt() == '1':
        break
    
    
N=rp_s.txrx_txt('ACQ:TPOS?')
print(N)
rp_s.tx_txt('ACQ:SOUR1:DATA?')
buff_string = rp_s.rx_txt()
buff_string = buff_string.strip('{}\n\r').replace("  ", "").split(',')
buff = list(map(float, buff_string))

rp_s.tx_txt('ACQ:STOP')

x=np.arange(0,len(buff),1)
plot.scatter(x,buff)
plot.ylabel('Voltage')
plot.show()`

In addition to that the trigger event position (labeled N in my code) returns very different numbers each time, even though the delay for the trigger is set to 0.  

In contrast, when using ACQ:TRig NOW, instead of the CH1 trigger with a level, the acquisition process and data plot occurs as expected, as seen in the following picture:

[image]

Finally, another thing that I noticed is that the buffer doesn't appear to be clearing properly with the ACQ:RST command because after I delete the variables where the data are stored in the computer memory and run again this part of the code:

`rp_s.tx_txt('ACQ:SOUR1:DATA?')
buff_string = rp_s.rx_txt()
buff_string = buff_string.strip('{}\n\r').replace("  ", "").split(',')
buff = list(map(float, buff_string))

rp_s.tx_txt('ACQ:STOP')

x=np.arange(0,len(buff),1)
plot.scatter(x,buff)
plot.ylabel('Voltage')
plot.show()`

I still get the same plot as before. It would be helpful if someone can tell me a reliable way of clearing the buffer.

b-swat, 2024-04-12:
Hello. There is no logic for clearing the buffer.

`rp_s.tx_txt('ACQ:TRig:DLY 0')`

Check the doc
[https://redpitaya.readthedocs.io/en/latest/appsFeatures/remoteControl/command_list.html#acquisition-trigger](https://redpitaya.readthedocs.io/en/latest/appsFeatures/remoteControl/command_list.html#acquisition-trigger)

This value indicates that only half of the buffer after the middle will be written, as can be seen in the picture.

Use case:
 // Start write data in buffer (If you give a zero signal, the buffer will be filled with zeros)
 rp_s.tx_txt('ACQ:START')
// wait for fill buffer before trigger
sleep(N)
// start trigger logic
rp_s.tx_txt('ACQ:TRig NOW')
// check the trigger has arrived

while 1:
rp_s.tx_txt('ACQ:TRIG:STAT?')
if rp_s.rx_txt() == 'TD':
break

// checking that all the data has been written

while 1:
rp_s.tx_txt('ACQ:TRIG:FILL?')
if rp_s.rx_txt() == '1':
break

// it's just a pointer in a cyclic buffer where data is written to the FPGA
// it will always randomly be greater than equal to 0 and less than the buffer size (16 * 1024)
N=rp_s.txrx_txt('ACQ:TPOS?')
print(N)

edgar-bonet, 2024-04-12:
@b-swat: The documentation does not say that “only half of the buffer after the middle will be written”. It says “the triggering moment is by default in the middle of acquired buffer”. From the docs, one could reasonably expect the trigger to only be armed once the buffer is half full.

Maybe a general explanation on how the trigger works could be helpful in that documentation page. As it stands, that documentation does not give the “big picture”: only a very terse explanation of each API/SCPI call.

b-swat, 2024-04-12:
The code does not check whether half of the buffer is full. This check must be done by the user.
PS. SCPI does not have this command, but the API does have rp_AcqGetPreTriggerCounter. Will have to add it in the future.
There is also a command in the api: rp_AcqSetTriggerDelayDirect which ignores the logic of half the buffer, it is also not yet available in SCPI

pjcoria, 2024-04-16:
Thank you for your answers! The thing that helped the most is the fact that when he trigger delay is set to 0 the acquisition post trigger starts in the middle of the buffer. The trigger was always working properly but I was failing to see that the signal I wanted to acquire was from sample number 8192 (middle of the buffer) onwards. When I added a vertical line there it was easier to spot that everything before that point was "old" data that I can discard:

[image]

In this case I was triggering with a pulse signal in CH2 instead of the same channel I wanted to measure, as I did before. This closer resembles the type of data acquisition I will be doing with the Red Pitaya:

 
[image]

The code I used is the following:

`
import sys
import redpitaya_scpi as scpi
import matplotlib.pyplot as plot
import numpy as np
import time as time

IP = 'rp-f04df7.local'

rp_s = scpi.scpi(IP)

rp_s.tx_txt('ACQ:RST')

rp_s.tx_txt('ACQ:DATA:FORMAT ASCII')
rp_s.tx_txt('ACQ:DATA:Units VOLTS')

rp_s.tx_txt('ACQ:DEC 32')

rp_s.tx_txt('ACQ:TRig:LEV 0.5 V')
rp_s.tx_txt('ACQ:TRig:DLY:NS 0')

rp_s.tx_txt('ACQ:START')
time.sleep(0.0)
rp_s.tx_txt('ACQ:TRig CH2_PE')

while 1:
    rp_s.tx_txt('ACQ:TRIG:STAT?')
    if rp_s.rx_txt() == 'TD':
        break
while 1:
    rp_s.tx_txt('ACQ:TRIG:FILL?')
    if rp_s.rx_txt() == '1':
        break
    
    
rp_s.tx_txt('ACQ:SOUR1:DATA?')
#rp_s.tx_txt('ACQ:SOUR1:DATA:Old:N? 3000')
buff_string = rp_s.rx_txt()
buff_string = buff_string.strip('{}\n\r').replace("  ", "").split(',')
buff = list(map(float, buff_string))

rp_s.tx_txt('ACQ:STOP')

x=np.arange(0,len(buff),1)
plot.plot(x,buff)
plot.ylabel('Voltage')
plot.grid()
plot.axvline(8192, color='red')
plot.show()
`

Note that I included a time.sleep as b-swat suggested and that when set to a proper amount of time did help with removing garbage data before the trigger. In my current code is set to 0 because I learned that the data before the trigger is not something I will be needing. 

Before solving my problem I tried using the SCPI command ACQ:SOUR1:DATA:Old:N? that can be found in the documentation. It allegedly reads a specified amount of samples after the trigger delay ( in my case it should read the data after the sample 8192), but it never worked as intended, at least for me. It always returned different data and away from the trigger. Has anyone used this successfully and can explain how this works exactly? or is this command just broken for the time being?. Another thing that I couldn't figure out is the data pointers. When I have the acquisition working as it should the problem I noticed is that when I ask for the position of the trigger event I always get a different result, even though I know that is triggering in sample number 8192. Something similar happens with the write pointer, it appears to return random numbers and never be at the end of the buffer where it should be when you have filled the buffer.

b-swat, 2024-04-16:
It seems that the documentation is not written entirely transparently. The old command increments the data “before the trigger.” try last command

b-swat, 2024-04-16:
I don't really like these "last" and "old" commands. I don't use them in development.

ACQ:SOUR<n>:DATA:STArt:End?
as parameters, specify the position of the trigger and the amount of data after it

edgar-bonet, 2024-04-16:
@pjcoria wrote:
> Another thing that I couldn't figure out is the data pointers.

The acquired data is stored in a [circular buffer][1]. The [documentation on the data pointers][2] states:

> The data is written into a circular buffer, which is constantly overwritten, until the triggering moment. Consequently, the trigger position can be anywhere inside the circular buffer, even though it is displayed to happen at approximately 8192nd sample in the acquired data [...]

My understanding is that the commands that return “pointer positions” really return actual raw positions within the buffer, not the positions that appear in the “unwrapped” arrays returned by the “data read” commands.

[1]: https://en.wikipedia.org/wiki/Circular_buffer
[2]: https://redpitaya.readthedocs.io/en/latest/appsFeatures/remoteControl/command_list.html#data-pointers

closed 2024-05-23 by b-swat