RedPitaya/RedPitaya #285: Failed to start OpenBSD Secure Shell server

fpgags, 2024-04-12:
Hi,
This is my first time using the board and updating the firmware version to RedPitaya-OS_2.00-30_stable. An error occurred during startup and kept restarting. How can I solve this problem?
[image]
I tried updating the firmware to an older version but it still couldn't function properly. Before version 0.99, it can start normally but cannot be accessed on the web.PC can ping successfully, but web access will crash directly.

fpgags, 2024-04-12:
STEMlab_125-xx_OS_0.99-41_beta
Unable to access web. The system will crash and display Getting the list of applications.
[image]
[image]
[image]
[image]

b-swat, 2024-04-12:
Hi
The problem with OpenBSD Secure Shell server is known to me.
I don't know how to fix it yet. The server does not start because the SSH service takes a long time to create certificates.
This happens on the first startup after writing to the SD card.
This does not cause any discomfort. ssh is available some time after the certificates are created.

And the problem with rebooting is not related to the OpenBSD server.
This bug is known to me; it was also fixed in build 35.
Problem with incorrect uboot initialization.
On some boards 30 build may not work.

fpgags, 2024-04-13:
Thank you for your answer

closed 2024-07-15 by b-swat