RedPitaya/RedPitaya #286: RedPitaya STEMlab125-14 problem

oligofrenia, 2024-04-12:
Hello, 
I am an owner of RedPitaya STEMlab 125-14. I am trying to use my device for deep memory signal acquisition. I tried to use example code from the documentary(https://redpitaya.readthedocs.io/en/latest/appsFeatures/examples/DMA/deepMemoryAcq.html). I decided to use the Python code version. When I try to use this code, I have an error "ModuleNotFoundError: No module named 'rp'". Unfortunately, The path "/opt/redpitaya/lib/python/" does not exist on my device. I have  "/opt/redpitaya/lib/" path, but "python/"  folder does not exist there, so I cant add this to my .bashrc file(https://redpitaya.readthedocs.io/en/latest/appsFeatures/remoteControl/API_scripts.html#running-python-applications). What should I do next? Is there any possibility of downloading this library from your repository (I don't see this in the RedPitaya project on GitHub)?

b-swat, 2024-04-12:
Hello. The documentation corresponds to the latest version of the release. We do not support documentation for older versions.
All updates are downloaded here.
https://downloads.redpitaya.com/downloads/Unify/

oligofrenia, 2024-04-12:
Can I use OS 2.0 version on my device? I am not sure if that device is not too old.

b-swat, 2024-04-12:
We have full support for older devices starting from 125-10. At least we test on all devices)))

oligofrenia, 2024-04-15:
Thanks for your support. I downloaded a *.img file with OS 2.0 and flashed my SD card. I ran data acquisition, and everything was fine, but after I turned it off, and tried to turn it on again, my device did not work. I reflashed my SD card, but I still can't connect with rp-xxxxx.local address

oligofrenia, 2024-04-16:
Hey, 
I loaded  1.04-28 ecosystem and I found that my hw_rev is correct: hw_rev=STEM_125-14_v1.0. After that, I tried to connect by serial port by PuTTY, and I had access to my device. It looks like SSH and web-based view not working and something is wrong with the network tools.
Best regards, 
Oliwia

b-swat, 2024-04-16:
check on another SD card

oligofrenia, 2024-04-16:
I checked and I have exactly the same situation. OS 1.04 work and 2.0 does not

b-swat, 2024-04-16:
It's hard to say that. you need to look at the systemd log. but looks like a certificate creation problem. Does RP have internet access?

oligofrenia, 2024-04-16:
I am working now only with LAN connection without internet access.

b-swat, 2024-04-16:
Try connecting to the Internet to create an SSL certificate, after which you can work on the Internet without it.

oligofrenia, 2024-04-17:
Okey, it was a problem with ssh key. Now everything fine :) . Finally, I have one more question: do you know what driver instruction to collect the entire buffer and save it in the background to the SD card or redpitaya?

oligofrenia, 2024-04-17:
Ok, It was problem with ssh key, everything fine now. :)

closed 2024-04-17 by oligofrenia