RedPitaya/RedPitaya #287: DMA questions

oligofrenia, 2024-04-18:
Hi,
I am trying to use my device for data acquisition. I need to acquire samples from RF-IN1 by 2 minutes with a 256 decimation factor (around 60 mln samples). I suppose that Deep Memory Acquisition is the solution to this problem. Is there any possibility to write buffer to file and use it again without sample loss and without using multi-threading? 

I any possibility of saving the file on an SD card near the system? In the final version, I need 10 minutes of acquisition and I am worried about the memory.

edgar-bonet, 2024-04-18:
At 256 decimation factor, you don't need Deep Memory Acquisition. The default 16K buffer can last for 33.55&nbsp;ms. You then have plenty of time to read the buffer and write the data to the SD card. Basically you stream the data like this:

    for 2 minutes:
        read the current position of the write pointer
        read the samples that were acquired since last time
        write those samples to the SD card

oligofrenia, 2024-04-18:
Is there any possibility of running a stream-server in a terminal without rpsa_client? 
I need a data acquisition tool that will run a 1 minute after turning on the power, collect measurements for 10 minutes, and save them to a file. The important thing is that the program can be run by a bash script or something like that without any external interference and without an Internet connection. Only after collecting the data do I want to connect the device and download the data file. As for the stream-app, I didn't see an example in the documentation, nor was I able to run the example from the terminal on the device itself.

b-swat, 2024-04-19:
Theoretically, this can be done. But I didn't check))
In the folder /opt/redpitaya/sbin
there is a script
https://github.com/RedPitaya/RedPitaya/blob/master/patches/startup/startup.sh
You can add the launch of the streaming server to it at the end.
and after a minute launch the streaming client in control mode. The client itself will not capture data, but will only start the server.

Or you can start the client for 10 minutes a minute after the server starts. This can all be done via bash scripts.

It is necessary to configure it before starting the server. A description of the setting should be in the documentation.

Starting the server from the console is described here
https://github.com/RedPitaya/RedPitaya/blob/master/apps-tools/streaming_manager/src/streaming-server/streaming-server.run.sh

oligofrenia, 2024-04-19:
How could I use rpsa client on my Red Pitaya board? I understand I need to use a client in parallel with a server on a device. To sum up: 
1. I need to run a stream-app using streaming-server.run.sh (https://github.com/RedPitaya/RedPitaya/blob/master/apps-tools/streaming_manager/src/streaming-server/streaming-server.run.sh) with appropriate configuration in the config file. 
2. I need to run a rpsa-client after a 1 minute. Rspa-client needs some arguments (eg. rpsa_client.exe -h 192.168.1.29 -p TCP -f ./ -t wav). I understand I can use it in local mode (not in TCP mode).

So to most important question now is how to run rpsa_client on the RP board and how to find information on what arguments to provide when starting the client.

b-swat, 2024-04-19:
I forgot that there is no pre-built client for RP.
It can be compiled from sources on the RP itself
It also has a large command reference

b-swat, 2024-04-19:
That's why I said theoretical))
I didn't support building the client for the RP itself. But it is possible to correct these errors.

oligofrenia, 2024-04-22:
Ok, I have done that, now everything works correctly. Thanks for your support and advice!

closed 2024-04-22 by oligofrenia