RedPitaya/RedPitaya #288: Trigger mode NORMAL not working in latest scope app of OS 2.00-35

HB9GKW, 2024-04-20:
By using trigger mode NORMAL the scope stays in mode TRIG'D forever once a trigger condition has happened. Expected would be to return to WAITING, if the trigger condition is not met anymore. Effectively this puts the scope in AUTO mode once a single trigger has happened and makes the scope useless for many applications.

HB9GKW, 2024-04-20:
after rebooting I cannot reproduce this bug..

HB9GKW, 2024-04-20:
Totally weird! It depends on time scale setting. To reproduce generate a simple 100 Hz square wave with 1 V amplitude on OUT1 and connect to IN1. Adjust time scale to 2 ms / div and put trigger mode to NORMAL with input to IN1. Everything looks as expected and trigger state jumps to WAITING as soon as I disconnect the cable.
At 5 ms / div the IN1 gets inverted. 
At 10 ms / div everything fine again. 
At 20 ms / div it completely goofs up by showing unstable readout of IN1 and trigger state jams at TRIG'D even when disconnecting the cable.

Lightsaver7, 2024-04-24:
Hello HB9GKW,
Thank you for reporting this. I will try to reproduce it and inform the developers.

Lightsaver7, 2024-04-24:
BTW what Red Pitaya model are you using? STEMlab 125-14?

HB9GKW, 2024-04-24:
Thanks for confirming. I could reproduce this bug on STEMlab 125-14 and STEMlab 125-14 4-channel input.

b-swat, 2024-05-22:
@HB9GKW 
The issue should be fixed in the latest nightly build

closed 2024-07-15 by b-swat