RedPitaya/RedPitaya #289: BIN file parse

oligofrenia, 2024-04-24:
Hi, 
I have a problem with getting data from the BIN file. I acquire data using a streaming server. I'm trying to get up 100000 samples. In the WAV file, I have exactly 100000 samples, but when I try to get data from the BIN file I have 100329 samples with a weird waveform. I think, it can be any header or something like that in this file but I do not find information about the structure in this file in the documentation. 
[image]

Lightsaver7, 2024-04-24:
Hello,
Thank you for letting us know. I will consult with the team regarding this and modify the docs to include this information.
If I understand correctly, you are streaming data from the Streaming application to the SD card?

oligofrenia, 2024-04-24:
Yes, I use the Data Stream Control app from the web interface. I save the file locally (to the SD card).

Lightsaver7, 2024-04-24:
Thanks b-swat, I'll add this to docs :D

oligofrenia, 2024-04-24:
Great, everything works perfectly!

closed 2024-04-24 by oligofrenia