RedPitaya/RedPitaya #290: /opt/redpitaya/sbin/startup.sh file edition

oligofrenia, 2024-04-24:
Hi, 
I try to edit the startup.sh (/opt/redpitaya/sbin/startup.sh) file. Is this possible? There is a phrase: "# Here you can specify commands for autorun at system startup", but when I try to edit this file, I see that the file is unwritten. How could I fix it?

b-swat, 2024-04-24:
The opt folder is read-only by default. in order to enable recording you need to run the command “rw” in the console

oligofrenia, 2024-04-25:
I did it. I can edit the start.sh script. Unfortunately, I doesnt matter what I write to this script, it does not work. Do I need to do anything else besides editing the script? I try to run bash script on boot. (init.d ans systemctl doesnt work too)

b-swat, 2024-04-26:
The script is launched via systemd, most likely it prohibits the execution of non-forked programs.
Well, these are the features of Linux

closed 2024-04-26 by oligofrenia