RedPitaya/RedPitaya #291: Please enable API access for additional IOs DIO8_P/N - DIO10_P/N

HB9GKW, 2024-05-01:
STEMlab version 125-14 4-input / 125-14 LN 7020 and SDRlab 122-16 do offer additional IOs that are currently not supported by the C API. Please make them available.

Lightsaver7, 2024-05-08:
Thank you for reporting this. We have added it to the development list and it will be fixed.

Lightsaver7, 2024-05-13:
Correction. The script above does not work for DIOP pins. Use this example for both DION and DIOP pins for now:
```

#include <stdio.h>
#include <stdlib.h>
#include "rp.h"

int main (int argc, char **argv) {
    uint32_t tmp;
    uint32_t state = 1;

    // Initialization of API
    if (rp_Init() != RP_OK) {
        fprintf(stderr, "Red Pitaya API init failed!\n");
        return EXIT_FAILURE;
    }

    // configure DIO[0:7]_N to outputs
    for (int pin=8; pin<11; pin++) {
        //rp_DpinSetDirection (i+RP_DIO0_P, RP_OUT);

        rp_GPIOpGetDirection(&tmp);
        rp_GPIOpSetDirection((tmp & ~(1 << pin)) | ((RP_OUT << pin) & (1 << pin)));

    }

    // configure DIO[0:7]_N to 1
    for (int pin=8; pin<11; pin++) {
        //rp_DpinSetState (i+RP_DIO0_P, 1);

        rp_GPIOpGetState(&tmp);
        rp_GPIOpSetState((tmp & ~(1 << pin)) | ((state << pin) & (1 << pin)));
    }

    // Releasing resources
    rp_Release();

    return EXIT_SUCCESS;
}
```

b-swat, 2024-05-22:
@HB9GKW 
Check last build #299
https://downloads.redpitaya.com/downloads/Unify/nightly_builds/
I've added support for additional pins.

Lightsaver7, 2024-05-24:
Confirmed functionality on Nightly Build 299 with this code for both P and N pins:
```
// configure DIO[8:11]_P to outputs
    for (int i=8; i<11; i++) {
        rp_DpinSetDirection (i+RP_DIO0_P, RP_OUT);
    }

    // configure DIO[8:11]_P to 1
    for (int i=8; i<11; i++) {
        rp_DpinSetState (i+RP_DIO0_P, 1);
    }
```
We can close this issue.

closed 2024-07-15 by b-swat