RedPitaya/RedPitaya #293: strange arbitrary output.

karu2003, 2024-05-16:
I'm getting a very strange output signal.
https://github.com/karu2003/tester_RedPitaya/blob/master/check_sequence.py
[image]

I can't upload it to RedPitaya.

Andrew.

b-swat, 2024-05-20:
Hello. I looked at the code, I'm not sure what needs to be done, but it seems to me that the buffer is not filled correctly.
For the signal to be correct, the 16k buffer must be filled with 1 signal period, and the frequency of signal generation is changed using the Freq command.
If you write a signal of a higher frequency (2.3, etc. hertz) into the buffer, then these two mechanisms will overlap each other and there will be unexpected behavior.

karu2003, 2024-05-21:
the buffer contains the correct number of points 16k.  no matter what it's filled with.  one frequency, two or complex shape.  Taking into account dec, i can calculate the maximum period and set the frequency. Do I get it right?

b-swat, 2024-05-21:
Decimation is not always used there, a buffer step is used. If we assume there are two signals in the buffer, then the frequency generator will double, but the generation error will be greater.

Here is the step calculation code
uint32_t value = (uint32_t) round(65536 * frequency / baseFreq * DAC_BUFFER_SIZE);

What the oscilloscope shows is rather always jumping from one period to another. Although the points seem to be nearby according to the calculations, but due to errors in the calculations of both the step and the signal, these glitches occur.

A step of one sample across the buffer with a signal of 1 period in the buffer will produce a signal with a frequency of 7629Hz

karu2003, 2024-05-21:
frequency will the sample rate?

b-swat, 2024-05-21:
The frequency of the signal depends on the step at which the read pointer will move across the memory buffer.

karu2003, 2024-05-21:
maybe you have an example of how to correctly create a sequence of two signals?

b-swat, 2024-05-21:
It is impossible to create a sequence of two buffers. you will have to mix two signals into 16. and reduce the frequency by half

karu2003, 2024-05-21:
How come? I have a buffer. There is time for one step. You can define Sample Rate. This should be enough to generate a sequence of two or more frequencies in one buffer. This is how all arbitrary generators work. The signal duration is determined by the buffer length and step time. Which suits me, but it doesn’t work quite right in RedPitaya. The only thing I don't give. -  round.

b-swat, 2024-05-22:
I agree that the generator does not work entirely correctly; there were problems related to the length of the buffer and the step. But so far there have been no plans to remake this logic.

karu2003, 2024-05-22:
The documentation says that RedPitaya can AWG. It works with chirp, but not with two sine. :) :(

b-swat, 2024-05-22:
what settings and what data are you using?

karu2003, 2024-05-23:
in a similar way I generate a chirp and a sequence of two "sin". Chirp works I "sin" looks bad.

b-swat, 2024-05-23:
I need a file with data for two sines and settings so that I can look at it myself

karu2003, 2024-05-31:
def gen_signals_sequence(frequencies, duration=0.0001, sample_rate=10e6):

Into https://github.com/karu2003/tester_RedPitaya/blob/master/signal_helper.py

b-swat, 2024-06-03:
Hello. I checked the signal.
)) I can say that the logic that is implemented in the generator is not suitable for creating this kind of signals.

Basic limitation: The buffer must contain a signal of one period (There may be several periods, but there must be enough data for the buffer step to build a periodic signal.) and can be of arbitrary shape.

The logic of the generator is that we lower the frequency of the generator by 65535 implicitly. To do this, we use the same divider and actually obtain a generator with a frequency of 2kHz.
And in order to generate a signal with a frequency of 1 Hz and a number of samples of 16k, we use step 8.
In order to receive a signal of 2 Hz, we already need step 16.

The logic of the generator is that we lower the frequency of the generator by 65535 implicitly. To do this, we use the same divider and actually obtain a generator with a frequency of 2kHz.
And in order to generate a signal with a frequency of 1 hertz and a number of samples of 16k, we use step 8.
In order to receive a signal of 2 hertz, we already need step 16.

I took the signal that you get in the file: check_sequence.py

>   frequencies = [250e3 , 500e3 ]
>   t, signal = sh.gen_signals_sequence(frequencies, duration=duration, sample_rate=sample_rate)

And I got this result.

[image]

This is closely similar to your problem.
With these settings, the buffer step is: 8590
which is not a multiple of 16384.
In fact, we 2 take two points from the buffer and then shift by the difference: 17180 - 16384 = 796 steps, thereby we get to the adjacent signal in the buffer, since we have many periods.

I can suggest this solution))
[image]

There were no plans to redo the generator logic yet.

karu2003, 2024-06-11:
Can I take a look at your implementation?

b-swat, 2024-06-14:
`#!/usr/bin/python3

import redpitaya_scpi as scpi
import sys
import math

rp_s = scpi.scpi("200.0.0.29")

rp_s.tx_txt('ACQ:BUF:SIZE?')
BUFF_SIZE = int(rp_s.rx_txt())

x = ''
t = []
half_buff = int(BUFF_SIZE / 2)
for i in range(0, half_buff):
	v = math.sin(4 * i * math.pi / (BUFF_SIZE/2))
	x += str(v) + ","

for i in range(0, half_buff):
	v = math.sin(2 * i * math.pi / (BUFF_SIZE/2))
	x += str(v)
	if (i < half_buff - 1):
		x += ','

print(x)

rp_s.tx_txt('GEN:RST')

rp_s.tx_txt('SOUR1:FUNC ARBITRARY')

rp_s.tx_txt('SOUR1:TRAC:DATA:DATA ' + x)

rp_s.tx_txt('SOUR1:VOLT 1')

rp_s.tx_txt('SOUR1:FREQ:FIX 4000')

rp_s.tx_txt('OUTPUT:STATE ON')
rp_s.tx_txt('SOUR:TRIG:INT')

#print x

rp_s.tx_txt('SOUR1:TRAC:DATA:DATA?')
buff_string = rp_s.rx_txt()
buff_string = buff_string.strip('{}\n\r').replace("  ", "").split(',')
buff = list(map(float, buff_string))

import matplotlib.pyplot as plt
plt.plot(buff)
plt.ylabel('Voltage')
plt.show()
`

b-swat, 2024-06-14:
In the screenshots I used the ARB manager
https://redpitaya.readthedocs.io/en/latest/appsFeatures/applications/arb_manager/arb_manager.html

closed 2024-07-15 by b-swat