RedPitaya/RedPitaya #295: Error messages on rp_Init()

edgar-bonet, 2024-06-25:
On a “STEM 125-14 4-Input” board, calling the `rp_Init()` function from the C API produces eight error messages on `stderr`.

Test program:

```c
#include <stdio.h>
#include <rp.h>

int main(void) {
    int err = rp_Init();
    printf("rp_Init() returned %d\n", err);
}
```

Output:

```text
[Error:acq_axi_SetBufferSamples] The specified buffer size is greater than the reserved memory - 8 bytes. End address: 0x0 End reserved 0x0
[Error:acq_axi_SetBufferSamples] The specified buffer size is greater than the reserved memory - 8 bytes. End address: 0x0 End reserved 0x0
[Error:acq_SetGain] Error setting threshold
[Error:acq_SetGain] Error setting threshold hysteresis
[Error:acq_axi_SetBufferSamples] The specified buffer size is greater than the reserved memory - 8 bytes. End address: 0x0 End reserved 0x0
[Error:acq_SetGain] Error setting threshold
[Error:acq_SetGain] Error setting threshold hysteresis
[Error:acq_axi_SetBufferSamples] The specified buffer size is greater than the reserved memory - 8 bytes. End address: 0x0 End reserved 0x0
rp_Init() returned 0
```

The initialization seems nevertheless successful, although so far I only tested the slow analog outputs.

System information (extracted from `/tmp/sysinfo.json`):

```yaml
model: STEM_125-14_Z7020_4IN_v1.3
mac: 00:26:32:F0:B4:6D
dna: 0044ccc01d09685c
ecosystem:
  name: Ecosystem
  version: 2.00-30
  revision: 73ff4bc8f
  description: Red Pitaya GNU/Linux OS & Ecosystem.
  linux_ver: '2.03'
linux: '2.03'
mem_size: '461'
mem_upgrade: '0'
boot_512: '1'
fpga:
  barebones: 6b685d6b3
  logic: 6b685d6b3
  stream_app: 6b685d6b3
  v0.94: 6b685d6b3
```

edgar-bonet, 2024-06-25:
I found where the first error message came from. The call chain is: `rp_Init()` → `rp_InitReset(true)` → `rp_Reset()` → `rp_AcqReset()` → `acq_Reset()` → `acq_SetDefault()` → `acq_axi_SetBufferBytes((rp_channel_t)i, start, 0)` → `acq_axi_SetBufferSamples(channel, address, 0)`.

At this point, `acq_axi_SetBufferSamples()` is called with `_samples = 0`. However, this function contains the following code:

```c
if (address + (_samples - 4) * 2 > start + res_size){
    ERROR("The specified buffer size is greater than the reserved memory - 8 bytes. End address: 0x%X End reserved 0x%X",address + _samples * 2,start + res_size);
    return RP_EOOR;
}
```

Given that `_samples` is of type `uint32_t` (a.k.a. `unsigned int` on ILP32), the subtraction `_samples - 4` is computed in this type. Being unsigned, it rolls over to a very large number, namely `UINT32_MAX - 3`.

I do not understand why 4 is being subtracted from `_samples`. However, it looks like the rollover is unintended, and the source of the unexpected error message.

b-swat, 2024-06-26:
Hi. I fix this function.
The value -4 appeared due to the operation of the FPGA DMA pipeline.
If you specify the beginning 0 and the end 100 of the memory region, then the FPGA will write data to addresses 104.
They didn’t fix this at the FPGA level, they just added a quick fix.

edgar-bonet, 2024-06-26:
Hi @b-swat, and thanks for your response.

Will that next release also fix the two other error messages (“Error setting threshold” and “Error setting threshold hysteresis”)?

b-swat, 2024-06-27:
yes

b-swat, 2024-07-03:
I checked the latest build from: https://downloads.redpitaya.com/downloads/Unify/nightly_builds/
It seems that I have already corrected these errors in the working version a long time ago.
Fixed the problem by returning an error code during initialization.
I set a restriction so that there would be no negative values ​​in the unsigned type.

edgar-bonet, 2024-07-16:
Hi @b-swat, and thanks for having fixed the problem!

Now, could you please leave the issue open until the fix is released?

Red Pitaya defines itself as an **open-source platform** (it's written in bold on its home page), and for some users, like me, this is one of its big selling points. It is thus pretty annoying to see issues being closed under the rationale that the problem is fixed… but only in a **closed-source** version of the platform!

Also, this is an invitation for users to post duplicate issues, as the GitHub repo and the GitHub issues are now out of sync.

edgar-bonet, 2024-07-16:
Thanks! Could you please also re-open #291? I see no occurrence of the string `RP_DIO9_N` in the whole repo.

edgar-bonet, 2025-12-04:
The fix is now released in Release-2025.2. Thanks!

closed 2025-12-04 by edgar-bonet