RedPitaya/RedPitaya #297: Please, make the acquisition state select()-able

edgar-bonet, 2024-06-28:
(If it already is, please provide a link to the relevant documentation.)

**Context**: I am tasked with building a firmware for a scientific instrument that will have a Red Pitaya as its brains. I plan to built it on top of the provided C API, using an event-driven, asynchronous, single-threaded architecture.

The provided example [acquire\_trigger\_posedge.c][example] waits for the acquisition to complete using busy-wait code equivalent to this:

```c
rp_acq_trig_state_t triggerState;
do {
    rp_AcqGetTriggerState(&triggerState);
} while (triggerState != RP_TRIG_STATE_TRIGGERED);

bool fillState;
do {
    rp_AcqGetBufferFillState(&fillState);
} while (!fillState);
```

This approach completely blocks the CPU while waiting, which makes it unsuitable for an event-driven program that may have other events to attend. In order to avoid this issue, would it be possible to make the state of the acquisition state machine (trigger state and fill state) accessible to [`select()`][select]/[`poll()`][poll]/[epoll][]?

My understanding is that the Red Pitaya C API is built on top of the [uio\_pdrv\_genirq][uio] generic uio kernel driver. This driver handles interrupts from the uio device, and makes them available to user space:

> You can also use **select()** on `/dev/uioX` to wait for an interrupt.

Then, in order to `select()` on the acquisition state machine, the user would only need:

1. A way to configure the interrupts sent by the FPGA. It would be nice if one could choose for the interrupt to be delivered when the acquisition is triggered **and** the buffer is full.

2. A way to get the uio file descriptor held by the library.

Is this possible? Planned? Could it be done?

[example]: ../blob/2.00-35/Examples/C/acquire_trigger_posedge.c
[select]: https://man7.org/linux/man-pages/man2/select.2.html
[poll]: https://man7.org/linux/man-pages/man2/poll.2.html
[epoll]: https://man7.org/linux/man-pages/man7/epoll.7.html
[uio]: https://www.kernel.org/doc/html/latest/driver-api/uio-howto.html#using-uio-pdrv-genirq-for-platform-devices

edgar-bonet, 2024-07-18:
While browsing through this repo, I noticed a piece of code that uses the uio file descriptor almost exactly in the manner I envision. [The method `COscilloscope::wait()`][code], defined in `apps-tools/streaming_manager/src/common_lib/uio_lib/oscilloscope.cpp`, calls `poll()` (a close cousin of `select()`) in order to wait for `POLLIN` events on the uio file descriptor `m_Fd`:

```c++
auto COscilloscope::wait() -> bool{
    int32_t cnt = 1;
    constexpr size_t cnt_size = sizeof(cnt);
    ssize_t bytes = write(m_Fd, &cnt, cnt_size); // Unmmask interrupt
    if (bytes == cnt_size) {
//      wait for the interrupt
//      printf("Wait Itr\n");
        struct pollfd pfd = {.fd = m_Fd, .events = POLLIN ,.revents = 0};
        int timeout_ms = 1000;
        int rv = poll(&pfd, 1, timeout_ms);
        // snip...
        return true;
    }
    return false;
}
```

This is almost what I would like to do, except that I would monitor multiple file descriptors, in order to respond to different kinds of events (as that's what `select()`/`poll()`/`epoll` are meant for).

Could this capability be exposed to the user?

[code]: ../blob/2.00-35/apps-tools/streaming_manager/src/common_lib/uio_lib/oscilloscope.cpp#L278-L303

b-swat, 2024-07-19:
Yes, we use interrupts there.
These are different FPGA projects and different implementations.
I asked the FPGA developer a question about adding this functionality. We will do this in the next release.