RedPitaya/RedPitaya #298: C API: how to switch an I/O pin from INPUT to OUTPUT HIGH with no glitch?

edgar-bonet, 2024-07-12:
I have to send an active-low logic signal to a external device. This device pulls its input HIGH with a resistor, in order to prevent it from reading LOW when it is not driven. When my program starts, I have to change the I/O pin state from INPUT (the default initial state) to OUTPUT HIGH, without going though an intermediate OUTPUT LOW state. How could I do this?

In the Arduino world, this would be as simple as:

```c++
digitalWrite(pin, HIGH);  // first set the desired level...
pinMode(pin, OUTPUT);     // ...then set the pin direction
```

I tried the same with librp:

```c
rp_DpinSetState(pin, RP_HIGH);     // first set the level
rp_DpinSetDirection(pin, RP_OUT);  // then the direction
```

This, however, does not work, as `rp_DpinSetState()` fails with `RP_EWIP`, which is documented as “Writing to Input Pin is not valid”.

If I set the direction before setting the state, the pin goes through an intermediate OUTPUT LOW state which I must avoid.

How can I go from INPUT to OUTPUT HIGH without this undesirable glitch?

edgar-bonet, 2024-08-29:
Gentle ping.

Is there a solution to this problem using the Red Pitaya library librp? Should I instead poke `/dev/gpiochip0` through [libgpiod][]? Wouldn't this interfere with an acquisition being done with librp (`rp_Acq*` functions)?

[libgpiod]: https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/about/

b-swat, 2024-09-03:
Check build #353
https://downloads.redpitaya.com/downloads/Unify/nightly_builds/ecosystem-2.05-353-3dd4ddea2.zip
and script:

```
#!/usr/bin/python3

import rp

print("rp.rp_InitAdresses()")
res = rp.rp_InitAdresses()
print(res)

print("rp.rp_DpinGetState(rp.RP_DIO0_P)")
res = rp.rp_DpinGetState(rp.RP_DIO0_P)
print(res)

print("rp.rp_DpinSetState(rp.RP_DIO0_P,rp.RP_HIGH)")
res = rp.rp_DpinSetState(rp.RP_DIO0_P,rp.RP_HIGH)
print(res)

print("rp.rp_DpinGetDirection(rp.RP_DIO0_P)")
res = rp.rp_DpinGetDirection(rp.RP_DIO0_P)
print(res)

print("rp.rp_DpinSetDirection(rp.RP_DIO0_P,rp.RP_OUT)")
res = rp.rp_DpinSetDirection(rp.RP_DIO0_P,rp.RP_OUT)
print(res)

print("rp.rp_Release()")
res = rp.rp_Release()
print(res)
```

I removed the mode check for GPIO.

edgar-bonet, 2024-09-03:
> Check build #353 [...]

I tried to sync the boot partition of the SD card with the contents of this archive, unsuccessfully. It turns out the archive contains a few pairs of files where the names differ by case only, such as `include/linux/include/linux/netfilter/{xt_DSCP.h,xt_dscp.h}`. The boot partition is FAT16, and it does not support such filenames differing by case only.

There are eight such problematic pairs. Since they are all in `include/linux/include/linux/`, which has not otherwise changed since the stable release RedPitaya\_OS\_2.04-35\_stable.img, I synced this specific directory of the SD card with the stable release, and the rest of the boot partition with the archive you provided. Hope this is OK.

> and [check this] script: [...]

It segfaulted on me:

```shell
$ ./test-gpio.py 
rp.rp_InitAdresses()
[E] {rp.c:rp_InitAdresses}(93) Error init osc regset. Code:  2
[E] {rp.c:rp_Release}(146) osc_Release() returned "Failed to close memory device."
2
rp.rp_DpinGetState(rp.RP_DIO0_P)
Segmentation fault
```

b-swat, 2024-09-03:
Maybe you miss load fpga?

b-swat, 2024-09-03:
What board are you testing on?

edgar-bonet, 2024-09-03:
> What board are you testing on?

125-14, 4 inputs. Sysinfo:

```json
{
  "model": "STEM_125-14_Z7020_4IN_v1.3",
  "mac": "00:26:32:F0:B4:6D",
  "dna": "0044ccc01d09685c",
  "ecosystem": {
    "name": "Ecosystem",
    "version": "2.05-353",
    "revision": "3dd4ddea2",
    "description": "Red Pitaya GNU/Linux OS & Ecosystem.",
    "linux_ver": "2.05"
  },
  "linux": "2.04",
  "mem_size": "461",
  "mem_upgrade": "0",
  "boot_512": "1",
  "fpga": {
    "barebones": "8f3f428a3",
    "logic": "8f3f428a3",
    "stream_app": "8f3f428a3",
    "v0.94": "8f3f428a3"
  }
}
```

I wrote:
> I'm going to try the C API.

If I try as the “redpitaya” user (with [access to the EEPROM][eeprom]), the program fails on `rp_Init()`, and segfaults if I ignore the error and try to `rp_DpinSetState()` anyway:

```text
$ ./gpio-test
[E] {rp.c:rp_InitAdresses}(93) Error init osc regset. Code:  2
[E] {rp.c:rp_Release}(146) osc_Release() returned "Failed to close memory device."
[E] {rp.c:rp_Init}(106) rp_InitAdresses() returned "Failed to open memory device."

rp_Init() -> 2
Segmentation fault
```

If I run the test as root, all calls return success. I haven't checked whether the pin behaves as expected.

Note that the failure of the Python script I reported was while running as the “redpitaya” user. Running as root gives:

```text
$ sudo ./test-gpio.py 
Traceback (most recent call last):
  File "/home/redpitaya/gpio/./test-gpio.py", line 3, in <module>
    import rp
ModuleNotFoundError: No module named 'rp'
```

<details><summary>My test program (click to expand/collapse)</summary>

```c
#include <stdio.h>
#include <rp.h>
#include <unistd.h>

const rp_dpin_t pin = RP_DIO0_P;

int main(void) {
    int err = rp_Init();
    printf("rp_Init() -> %d\n", err);
    err = rp_DpinSetState(pin, RP_HIGH);
    printf("rp_DpinSetState(pin, RP_HIGH) -> %d\n", err);
    rp_DpinSetDirection(pin, RP_OUT);
    printf("rp_DpinSetDirection(pin, RP_OUT) -> %d\n", err);
    sleep(1);
    err = rp_DpinSetState(pin, RP_LOW);
    printf("rp_DpinSetState(pin, RP_LOW) -> %d\n", err);
    rp_DpinSetDirection(pin, RP_IN);
    printf("rp_DpinSetDirection(pin, RP_IN) -> %d\n", err);
}
```

</details>

Now my plan is:

1. check on the scope whether the pin behaves as expected, when running the test as root
2. run as redpitaya and `strace` the test program to find out what permission it is missing
3. try to fix the permissions on the OS
4. report back here

[eeprom]: ../../ubuntu/pull/4

b-swat, 2024-09-03:
gpio works and is configured via FPGA registers.

b-swat, 2024-09-03:
https://redpitaya.readthedocs.io/en/latest/developerGuide/software/build/fpga/regset/2.00-35/v0.94.html#housekeeping

b-swat, 2024-09-03:
> Note that the failure of the Python script I reported was while running as the “redpitaya” user. Running as root gives:
In release 37 the rights for redpitaya user should be fixed.
https://github.com/RedPitaya/ubuntu/commits/2.05

> $ sudo ./test-gpio.py 
Traceback (most recent call last):
  File "/home/redpitaya/gpio/./test-gpio.py", line 3, in <module>
    import rp
ModuleNotFoundError: No module named 'rp'

The paths for the scripts are apparently not initialized, apparently for the redpitaya user.
https://github.com/RedPitaya/ubuntu/commit/a2f582da5c80718022f1c04beeb2cd8d3a35c741

> rp_Init()
This command, in addition to initializing access to registers, also resets all values ​​to default. Therefore, it should not be used in this example.

I specifically added a command that opens devices without resetting them.
rp_InitAdresses()

edgar-bonet, 2024-09-03:
> The paths for the scripts are apparently not initialized

`$PYTHONPATH` does contain `/opt/redpitaya/lib/python/`, both when logging in as the redpitaya user and when logging in as root (`sudo -i`). However, it seems the variable is not set when running a Python script through `sudo`.

> I specifically added a command that opens devices without resetting them.

I don't think this is relevant for me. After boot, the GPIO pins are in input mode.

------------------------------------------------------------------------
Now, as promised, I am reporting back my findings.

## Scope test

It works! :smile:

I connected a 1.5&nbsp;kΩ resistor between DIO0\_P and GND, and another 1.5&nbsp;kΩ resistor between DIO0\_P and +3V3. This way the pin stays at ≈&nbsp;1.7&nbsp;V when configured as INPUT. Then, with a scope, I can tell apart the three electric states INPUT, OUTPUT LOW and OUTPUT HIGH.

* If I set the direction to OUTPUT before setting the level to HIGH, there is a 2.2&nbsp;µs glitch in the OUTPUT LOW state.
* If I set the level to HIGH before setting the direction to OUTPUT, then the pin goes directly from INPUT to OUTPUT HIGH, with no glitch.

This is exactly what I was looking for. Thank-you! The remaining problem is that the program has to run as root.

## Permission problem

`strace` shows that the problem is that `rp_Init()` attempts to open `/dev/mem` read/write, which fails with `EACCES` (Permission denied).

I don't understand why `rp_Init()` is doing such a thing. Since commit c43bc892cfc6657cc46767cca3c25bac7ffd1c04, which dates back to 2017, the API accesses the FPGA through `/dev/uio/api` instead of `/dev/mem`. Why is this nightly reverting to `/dev/mem`?

## Permission fix

Both `/dev/mem` and the redpitaya user belong to the Unix group “kmem”, and the file is group-readable. It looks like making `/dev/mem` group-writable should fix the permission problem.

Alas, this doesn't work. Opening `/dev/mem` now fails with `EPERM` (Operation not permitted). Note that this is different from `EACCES` (Permission denied). It turns out [`/dev/mem` can only be opened by processes with `CAP_SYS_RAWIO`][se]. This is annoying, as it requires a `sudo setcap` after each build of any program using the API.

Is there a reason (other than oversight) that `rp_Init()` is attempting to open `dev/mem`? Should I report this as a separate issue?

[se]: https://unix.stackexchange.com/a/475807

b-swat, 2024-09-03:
I think the question is about this line in API.
https://github.com/RedPitaya/RedPitaya/blob/master/rp-api/api/src/oscilloscope.c#L81C9-L81C15

In this case, we open the device to access an arbitrary area of ​​memory in DMA mode.

If my memory serves me correctly, /dev/uio/api only has rights to a specific block of memory.

edgar-bonet, 2024-09-04:
> In this case, we open the device to access an arbitrary area of memory in DMA mode.

Oh, now I understand:

 * `/dev/uio/api` is used to access the FPGA, including its BRAM memory buffers used in regular data acquisition
 * `/dev/mem` is used to access the [reserved memory][reserved] that has been set apart for [Deep Memory Acquisition][dma] (`rp_AcqAxi*()` family of functions)

I though that `/dev/mem` access was a new thing in this nightly version. An `strace` test proved me wrong. It turns out the stable API (2.00-35) already opens `/dev/mem`, but the `EACCES`/`EPERM` errors are not visible in that version, because `rp_InitReset()` [does not check the return code of `osc_Init()`][ret].

What this implies that, for a user not using DMA, the API is needlessly accessing `/dev/mem`, which is annoying since it essentially forces the application to run as root.

One possible fix would be to only open `/dev/mem` when it's actually going to be accessed, namely [in `osc_axi_map()`][map], right before `mmap()`-ing it. I may propose a pull request implementing this.

Maybe another option could be to declare this memory as a “volatile” [persistent memory region][pmem], which would then be accessible as `/dev/pmem0`.

[reserved]: ../../RedPitaya-FPGA/blob/master/dts/memory.dtsi#L22-L24
[dma]: https://redpitaya.readthedocs.io/en/latest/appsFeatures/remoteControl/deepMemoryAcquisition.html
[ret]: ../blob/2.00-35/rp-api/api/src/rp.c#L64
[map]: ../blob/2.00-37/rp-api/api/src/oscilloscope.c#L947-L977
[pmem]: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/pmem/pmem-region.txt?h=v6.10

edgar-bonet, 2024-09-04:
I wrote:
> I may propose a pull request implementing this.

Done in #309.

edgar-bonet, 2025-12-04:
The problem is solved in Release-2025.2: I can get a scope trace with no glitch. I am therefore closing the issue.

Thank-you!

closed 2025-12-04 by edgar-bonet