RedPitaya/RedPitaya #300: Trigger is not arming in secondary device in multi channel acquisition through Daisy chain connection

hk06319, 2024-07-18:
Hi, I am trying to read the 4 channels on two STEMLab 125-14 boards triggered simultaneously through daisy chain. I have connected SATA S1 port of primary device to SATA S2 port of the secondary device with a SATA cable. I am using QS version 2.00-30 version on both devices. Here is the Python code I am running (similar to the one on redpitaya site):

import time
import matplotlib.pyplot as plt
import numpy as np

import redpitaya_scpi as scpi

# Connect OUT1 primary with IN1 primary and IN1 secondary

wave_form = "sine"
freq = 100000
ampl = 1

dec = 2
trig_lvl = 0.1
trig_dly = 8192

IP_PRIM = '10.107.54.172'   # IP Test OS Red Pitaya
IP_SEC = '10.107.54.171'

rp_prim = scpi.scpi(IP_PRIM)
rp_sec = scpi.scpi(IP_SEC)

print("Program Start")

rp_prim.tx_txt('GEN:RST')
rp_prim.tx_txt('ACQ:RST')
time.sleep(0.2)                           # Not necessary
rp_sec.tx_txt('GEN:RST')
rp_sec.tx_txt('ACQ:RST')
time.sleep(0.2)                           # Not necessary
###### ENABLING THE DAISY CHAIN PRIMARY UNIT ######

rp_prim.tx_txt('DAISY:SYNC:TRig ON')    #! OFF (without sync)
rp_prim.tx_txt('DAISY:SYNC:CLK ON')
#rp_prim.tx_txt('DAISY:TRIG_O:ENable ON')
rp_prim.tx_txt('DAISY:TRIG_O:SOUR ADC')

rp_prim.tx_txt('DIG:PIN LED5,1')            # LED Indicator

time.sleep(0.2)

print(f"Trig: {rp_prim.txrx_txt('DAISY:SYNC:TRig?')}")
print(f"CLK: {rp_prim.txrx_txt('DAISY:SYNC:CLK?')}")
print(f"Sour: {rp_prim.txrx_txt('DAISY:TRIG_O:SOUR?')}\n")
print(f"Trig Enable: {rp_prim.txrx_txt('DAISY:TRIG_O:ENable?')}\n")

###### ENABLING THE DAISY CHAIN SECONDARY UNIT ######

rp_sec.tx_txt('DAISY:SYNC:TRig ON')  #! OFF (without sync)
rp_sec.tx_txt('DAISY:SYNC:CLK ON')
rp_sec.tx_txt('DAISY:TRIG_O:SOUR ADC')     # Ext trigger will trigger the ADC
#rp_sec.tx_txt('DAISY:TRIG_O:ENable ON')

print(f"Trig: {rp_sec.txrx_txt('DAISY:SYNC:TRig?')}")
print(f"CLK: {rp_sec.txrx_txt('DAISY:SYNC:CLK?')}")
print(f"Sour: {rp_sec.txrx_txt('DAISY:TRIG_O:SOUR?')}\n")
print(f"Trig Enable: {rp_sec.txrx_txt('DAISY:TRIG_O:ENable?')}\n")

rp_sec.tx_txt('DIG:PIN LED5,1')             # LED Indicator

print("Start generator\n")

### Generation ### - Primary unit
rp_prim.tx_txt(f'SOUR1:FUNC {wave_form}')
rp_prim.tx_txt(f'SOUR1:FREQ:FIX {freq}')
rp_prim.tx_txt(f'SOUR1:VOLT {ampl}')

rp_prim.tx_txt('OUTPUT1:STATE ON')

### Aquisition ###

# Primary unit
rp_prim.tx_txt(f'ACQ:DEC {dec}')
rp_prim.tx_txt(f'ACQ:TRig:LEV {trig_lvl}')
rp_prim.tx_txt(f'ACQ:TRig:DLY {trig_dly}')

# Secondary unit
rp_sec.tx_txt(f'ACQ:DEC {dec}')
rp_sec.tx_txt(f'ACQ:TRig:LEV {trig_lvl}')
rp_sec.tx_txt(f'ACQ:TRig:DLY {trig_dly}')

rp_sec.tx_txt('ACQ:START')
time.sleep(0.05)                           # Not necessary
rp_sec.tx_txt('ACQ:TRig EXT_NE')          #! CH1_PE (without sync trig) EXT_NE (with sync trig)
                                          # If not synchronised make sure no signal arrives before both units are set                        

rp_prim.tx_txt('ACQ:START')
time.sleep(0.05)
rp_prim.tx_txt('ACQ:TRig Ch1_PE')

time.sleep(0.1)                             # Symulating a trigger after one second
rp_prim.tx_txt('SOUR1:TRig:INT')

print("ACQ start")

while 1:
    # Get Trigger Status
    if rp_prim.txrx_txt('ACQ:TRig:STAT?') == 'TD':               # Triggerd?
        break
print("Trigger primary condition met.")

## ! OS 2.00 or higher only ! ##
while 1:
    if rp_prim.txrx_txt('ACQ:TRig:FILL?') == '1':
        break
print("Buffer primary filled.") 

while 1:
    # Get Trigger Status
    if rp_sec.txrx_txt('ACQ:TRig:STAT?') == 'TD':               # Triggerd?
        break
print("Trigger secondary condition met.")

## ! OS 2.00 or higher only ! ##
while 1:
    if rp_sec.txrx_txt('ACQ:TRig:FILL?') == '1':
        break
print("Buffer secondary filled.")

# Read data and plot
rp_prim.tx_txt('ACQ:SOUR1:DATA?')               # Read full buffer primary (source 1)
data_string1 = rp_prim.rx_txt()                 # data into a string

rp_sec.tx_txt('ACQ:SOUR1:DATA?')                # Read full buffer secondary (source 1)
data_string2 = rp_sec.rx_txt()

# Display both buffers at once
n = 2
buff = np.zeros((n,16384))

# Remove brackets and empty spaces + string => float
data_string1 = data_string1.strip('{}\n\r').replace("  ", "").split(',')
data_string2 = data_string2.strip('{}\n\r').replace("  ", "").split(',')
# Transform data into data series
buff[0, :] = list(map(float, data_string1))
buff[1, :] = list(map(float, data_string2))

######## PLOTTING THE DATA #########

fig, axs = plt.subplots(n, sharex = True)               # plot the data (n subplots)
fig.suptitle("Measurements P1 S2")

for i in range(0,n,1):                                  # plotting the acquired buffers
    axs[i].plot(buff[i])

plt.show()

rp_prim.close()
rp_sec.close()
#################################################################
The output is 
>>>>>>>>
rogram Start
Trig: ON
CLK: 
Sour: ADC

Trig Enable: ON

Trig: ON
CLK: ON
Sour: ADC

Trig Enable: ON

Start generator

ACQ start
Trigger primary condition met.
Buffer primary filled.

I think secondary device is not triggered, I couldn't figure out why this is happening.

edgar-bonet, 2024-07-18:
Could you please edit your question and format your code? Use triple backtics with a “python” language identifier, as explained in this [markdown doc][doc].

[doc]: https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#syntax-highlighting

Lightsaver7, 2025-01-13:
Please try the code on the latest OS version of Red Pitaya - 2.05-37 with the example [here](https://redpitaya.readthedocs.io/en/latest/appsFeatures/examples/multiboard_sync/daisy_chain-exm1.html#daisy-chain-generation-and-acquisition)