RedPitaya/RedPitaya #302: very long signal at the generator output?

karu2003, 2024-08-14:
Is it possible to get a very long signal at the generator output?
something like this.
https://github.com/karu2003/tester_RedPitaya/blob/refactoring/evosonar-tests/18-34_test_sig.py

b-swat, 2024-08-15:
Maybe it will help, maybe not.
The streaming application has the ability to send a signal to the generator in stream mode.

DAC streaming Mode:
https://redpitaya.readthedocs.io/en/latest/appsFeatures/applications/streaming/appStreaming.html#instructions-for-the-rpsa-client

When I did it, it seemed to even work, but there were very few tests, since the functionality was not in demand by clients.
There you can transfer a wav file, which will be sent to the generator.

karu2003, 2024-08-15:
interesting. is it possible to use streaming from python or cli? are there any examples?

b-swat, 2024-08-15:
There is no client for Python, although we raised the question of the possibility of integration with Python. This mode is only available from the console. It cannot be configured via the web application.

In the latest version, the streaming server configuration file is located here:
/root/.config/redpitaya/apps/streaming/streaming_config.json

The file contains a section related to setting up DAC streaming:
	"dac_streaming" : 
	{
		"dac_file" : "",
		"dac_file_type" : 0,
		"dac_gain" : 0,
		"dac_memoryUsage" : 1048576,
		"dac_mode" : 0,
		"dac_port" : "8903",
		"dac_repeat" : -1,
		"dac_repeatCount" : 1,
		"dac_speed" : 125000000
	},

You can also configure the server through the console client.
```
rpsa_client.exe -c -h IPs [-p PORT] -g V|VV|F [-v]
rpsa_client.exe -c -h IPs [-p PORT] -s M|F [-f FILE] [-v]
rpsa_client.exe --config --hosts=IPs [--port=PORT] --get=V|VV|F [--verbose]
rpsa_client.exe --config --hosts=IPs [--port=PORT] --set=M|F [--config_file=FILE] [--verbose]

--config               -c           Enable config mode.
--hosts=IP,...         -h IP,...    You can specify one or more board IP addresses through a separator - ','
                                    Example: --hosts=127.0.0.1 or --hosts=127.0.0.1,127.0.0.2
                                              -h 127.0.0.1     or  -h 127.0.0.1,127.0.0.2,127.0.0.3
--port=PORT            -p PORT      Port for configuration server (Default: 8901).
--get=V|V1|VV|F        -g V|V1|VV|F Requests configurations from all boards.
                                    Keys: V  = Displays on the screen in json format.
                                          V1 = Displays on the screen in json format (only data).
                                          VV = Displays on the screen in a format with decoding values.
                                          F  = Saves to a config files.
--set=M|F              -s M|F       Sets configurations for all boards.
                                    Keys: M  = Sets values only to memory without saving to file.
                                          F  = Sets configuration and saves to file on remote boards.
--config_file=FILE     -f FILE      Configuration file for setting values on boards (Default: config.json).
--verbose              -v           Displays service information.
```

This client mode works exactly with this configuration file.

b-swat, 2024-08-15:
It was possible to transfer data for generation through the console client.

```
rpsa_client.exe -o -h IPs [-p PORT] [-c PORT] -f tdms|wav -d FILE_NAME [-r inf|COUNT] [-m SIZE] [-v] [-b]
rpsa_client.exe --out_streaming --hosts=IPs [--port=PORT] [--config_port=PORT] --format=tdms|wav --data=FILE_NAME [--repeat=inf|COUNT] [--memory SIZE] [--verbose] [--benchmark]
rpsa_client.exe -oc CONFIG_FILE
rpsa_client.exe --out_streaming_conf CONFIG_FILE

--out_streaming        -o           Enable dac streaming mode.
--out_streaming_conf   -oc          Enable dac streaming mode.
--hosts=IP,...         -h IP,...    You can specify one or more board IP addresses through a separator - ','
                                    Example: --hosts=127.0.0.1 or --hosts=127.0.0.1,127.0.0.2
                                              -h 127.0.0.1     or  -h 127.0.0.1,127.0.0.2,127.0.0.3
--port=PORT            -p PORT      Port for dac streaming server (Default: 8903).
--config_port=PORT     -c PORT      Port for configuration server (Default: 8901).
--format=FORMAT        -f FORMAT    The format in which the data will be used.
                                    Keys: tdsm = NI TDMS File Format.
                                          wav = Waveform Audio File Format.
--data=FILE_NAME       -d FILE_NAME Path to the file for streaming.
--memory=SIZE          -m SIZE      Use RAM cache.
                                        Example: --mmemory 1048576 or --memory 1M or --memory 1024k
                                                 -m 1048576 or -m 1M or -m 1024k
                                        Default: 1M bytes
--repeat=inf|COUNT     -r inf|COUNT The number of times the file has been resent.
                                    Keys: inf is an infinite number of times.
                                       COUNT - value from [1 ... 1000000]
--verbose              -v           Displays service information.
--benchmark            -b           Starts the throughput test mode at the current settings.
```

**--data=FILE_NAME       -d FILE_NAME Path to the file for streaming.**

It seems that you could also upload the file to the RP and specify the path to it in the configuration file. But I'm not sure that this mode worked.

b-swat, 2024-08-15:
Simple example for cleint: _rpsa_client.exe -o -h 200.0.0.100 -f wav -d 1.wav_

karu2003, 2024-08-15:
You work under Wine, I have Linux.  rpsa_client_qt.sh works without gui?

b-swat, 2024-08-15:
This needs to be debugged, I’m saying that it wasn’t tested enough.

karu2003, 2024-08-15:
Maybe he doesn't like this format? What should it be? float16..32 or so?

b-swat, 2024-08-15:
May be. Check data in  16 bit signed integer

karu2003, 2024-08-15:
on the RPI side I didn't turn on anything and started it ./rpsa_client -o -h 192.168.0.15 -r inf -f wav -d ~/src/tester_RedPitaya/1834_cs1.wav -v. on OUT i I see some kind of signal. This is already progress. But. It looks like a digital signal because it jumps back and forth.

karu2003, 2024-08-15:
How do you set up a sample rate for DAC?

karu2003, 2024-08-16:
I don't know how correct this is. I managed to figure out experimentally that in DAC mode you only have one sample rate? 125M. This is not always necessary. The mode can only be turned on in WEB Streaming App. It is necessary to go to the APP window. Even if nothing is started later. My WAV signal is very large 75M. It all works not very nicely. :(

b-swat, 2024-08-16:
Parameter "dac_speed" : 125000000
affects the generation frequency.

It is used here:
https://github.com/RedPitaya/RedPitaya/blob/master/apps-tools/streaming_manager/src/common_lib/uio_lib/generator.cpp#L155

I'm not sure it will help you anyway.
I simply suggested this solution as an option, which is at an early stage of development.
But now I don’t have time to figure out the bugs and complete this mode.

)) We also have plans to do GPIO streaming there. But when I do this I can't even imagine

karu2003, 2024-08-16:
no time - familiar to everyone. Who has it? :)  I'll try the brush mode. Maybe the delay between series will be acceptable. At the moment, there is no time to change the test platform. :)