RedPitaya/RedPitaya #304: GenGetLoadMode

FabricioS, 2024-08-28:
Hi,
I need to set the voltage range of the RF inputs/outputs of the SignalLab 250-12 from software (not using the jumper).
I found the Python-side **rp_GenGetLoadMode** function but it seems not to be functional:
```
res = rp.rp_GenGetLoadMode(rp.RP_CH_1)
print(res)
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In[3], line 1
----> 1 res = rp.rp_GenGetLoadMode(rp.RP_CH_1)
      2 print(res)

TypeError: rp_GenGetLoadMode() missing 1 required positional argument: 'mode'
```

Looking at the source code, it seems that function in fact expect an output argument to be provided (as in C interface where a pointer is to be provided). How can this be done from Python ?

Best regards
Fabrice

edgar-bonet, 2024-08-28:
According to [the documentation][doc], the required parameter is a channel number.

[doc]: https://redpitaya.readthedocs.io/en/latest/appsFeatures/remoteControl/command_list.html#generator-settings

b-swat, 2024-08-29:
Do you mean 1:1/1:20 for ADC and x1 and x5 for DAC?

FabricioS, 2024-08-29:
Indeed

FabricioS, 2024-08-29:
https://redpitaya.readthedocs.io/en/latest/developerGuide/hardware/250-12/top.html

**RF_inputs**:[...]
Full scale voltage range: +-1 V/+-20 V (software selectable)

**RF outputs**:[...]
Voltage range: +-2 V / +-10 V (Hi-Z load) (software selectable)

b-swat, 2024-08-29:
Check this
https://github.com/RedPitaya/RedPitaya/blob/master/rp-api/api/tests/rp_test.py

b-swat, 2024-08-29:
print("rp.rp_GenSetLoadMode(rp.RP_CH_1,rp.RP_GEN_50Ohm)")
res = rp.rp_GenSetLoadMode(rp.RP_CH_1,rp.RP_GEN_50Ohm)
print(res)

print("rp.rp_GenGetLoadMode(rp.RP_CH_1)")
res = rp.rp_GenGetLoadMode(rp.RP_CH_1)
print(res)

b-swat, 2024-08-29:
print("rp.rp_GenSetGainOut(rp.RP_CH_1,rp.RP_GAIN_1X)")
res = rp.rp_GenSetGainOut(rp.RP_CH_1,rp.RP_GAIN_1X)
print(res)

print("rp.rp_GenGetGainOut(rp.RP_CH_1)")
res = rp.rp_GenGetGainOut(rp.RP_CH_1)
print(res)

FabricioS, 2024-09-01:
Thank you b-swat,
Testing the commands you provided is almost OK:
```

Cmds = [
    (rp.rp_GenSetLoadMode, (rp.RP_CH_1,rp.RP_GEN_50Ohm)),
    (rp.rp_GenGetLoadMode, (rp.RP_CH_1,)),
    (rp.rp_GenSetGainOut,  (rp.RP_CH_1,rp.RP_GAIN_1X)),
    (rp.rp_GenGetGainOut,  (rp.RP_CH_1,)),
    (rp.rp_AcqSetGain,     (rp.RP_CH_1,rp.RP_HIGH)),
    (rp.rp_AcqGetGain,     (rp.RP_CH_1,))
]
Results = {}

for func, args in Cmds:
    try:
        Results[func] = res = func(*args)
    except:
        err = sys.exc_info()
        Results[func] = err[0]
        print(err[1])

print(Results)
```

gives :
```
rp_GenGetLoadMode() missing 1 required positional argument: 'mode'
rp_GenGetGainOut() missing 1 required positional argument: 'mode'

{<function rp.rp_GenSetLoadMode(channel, mode)>: 0,
 <function rp.rp_GenGetLoadMode(channel, mode)>: TypeError,
 <function rp.rp_GenSetGainOut(channel, mode)>: 0,
 <function rp.rp_GenGetGainOut(channel, mode)>: TypeError,
 <function rp.rp_AcqSetGain(channel, state)>: 0,
 <function rp.rp_AcqGetGain(channel)>: [0, 1]}
```

rp_GenGetLoadMode and rp_GenGetGainOut seem to have a poor wrapper that requires the user to provide a mode as an input argument (while these functions are suppose to output a mode or a gain).

FabricioS, 2024-09-01:
Note also thath `rp_GenArbWaveformNP` is mentioned in the test file you pointed, but not available in my installation.
I tried to check the version of the Python wrapper, unsuccessfully...
FYI, I upgraded the 2.00-35 release of the OS a few days ago

b-swat, 2024-09-02:
Check version 37.
They really don't work in 35. I guess I fixed this problem later.