RedPitaya/RedPitaya #305: Adquire one (or a fixed number) of samples after trigger event

gcorthey, 2024-08-29:
I need to adquire one sample or a fixed number of samples (say 10-100) after each trigger event and save to buffer.

For example, I have a trigger running at 1 KHz connected to channel 2. I start acquisition and I need to register 50 samples after each trigger event for 100 events and then read the buffer.

**How can I do that using scpi commands?**

b-swat, 2024-09-03:
Theoretically, you can capture in a cycle, but SCPI is not the fastest tool. So you can simply not have time to capture and miss the tigers

gcorthey, 2024-09-03:
Hello @b-swat thanks for your answer. Unfortunatelly I don't undersand you suggestion. Could you please expand it more?

b-swat, 2024-09-03:
Run data capture in a loop. As in this example:
https://github.com/RedPitaya/RedPitaya/blob/master/Examples/python/acquire_trigger_posedge.py

If the speed is not enough, then write an application in C++ or Python on the board itself.
Like here:
https://github.com/RedPitaya/RedPitaya/blob/master/Examples/C/acquire_trigger_software.c

https://github.com/RedPitaya/RedPitaya/blob/master/Examples/python-api/acq_trigger_level.py