RedPitaya/RedPitaya #306: Trouble with decimation factors and SCPI commands

pjcoria, 2024-08-30:
I'm having trouble changing the decimation factor using the SCPI command ACQ:DEC. It works fine with 1, 2, 4, 8, 16 and 32 but if I tell the RP to use a decimation factor greater than 32 (using powers of 2 of course) it just won't measure anything. I can get the setted decimation factor using ACQ:DEC? and it seems to be correct but when it is greater than 32 it seems to freeze the board's acquisition. Any ideas on what might be the problem or how to solve it?

b-swat, 2024-09-02:
Hello
Ecosystem version? example script?

pjcoria, 2024-09-03:
The ecosystem version is : [Red Pitaya OS 2.00-30 / STEMlab 125-14 v1.1]

I used a very simple script to measure using SCPI commands as follows: 

import sys
from Controller import redpitaya_scpi as scpi
import matplotlib.pyplot as plot
import numpy as np
import time as time

IP = 'rp-f04df7.local'

rp_s = scpi.scpi(IP)
#%%
rp_s.tx_txt('ACQ:RST')

rp_s.tx_txt('ACQ:DATA:FORMAT ASCII')
rp_s.tx_txt('ACQ:DATA:Units VOLTS')

rp_s.tx_txt('ACQ:DEC 64')

rp_s.tx_txt('ACQ:TRig:LEV 0.5 V')
rp_s.tx_txt('ACQ:TRig:DLY:NS 0')

rp_s.tx_txt('ACQ:START')
time.sleep(0.0)
rp_s.tx_txt('ACQ:TRig CH2_PE')
#rp_s.tx_txt('ACQ:TRig NOW')

while 1:
    rp_s.tx_txt('ACQ:TRIG:STAT?')
    if rp_s.rx_txt() == 'TD':
        break

## ! OS 2.00 or higher only ! ##
while 1:
    rp_s.tx_txt('ACQ:TRIG:FILL?')
    print(rp_s.rx_txt())
    rp_s.tx_txt('ACQ:TRIG:FILL?')
    if rp_s.rx_txt() == '1':
        break
    
    
rp_s.tx_txt('ACQ:SOUR1:DATA?')
#rp_s.tx_txt('ACQ:SOUR1:DATA:Old:N? 3000')
buff_string = rp_s.rx_txt()
buff_string = buff_string.strip('{}\n\r').replace("  ", "").split(',')
buff = list(map(float, buff_string))

rp_s.tx_txt('ACQ:STOP')

x=np.arange(0,len(buff),1)
plot.plot(x[8192:8900],buff[8192:8900])
plot.ylabel('Voltage')
plot.grid()
plot.axvline(8192, color='red')
plot.show()

b-swat, 2024-09-03:
Take an example from here?
https://github.com/RedPitaya/RedPitaya/blob/master/Examples/python/acquire_trigger_posedge.py

 I'm confused by the commands
There should not be V indicated there.
rp_s.tx_txt('ACQ:TRig:LEV 0.5 V')

There seems to be a mistake in the documentation.

[image]

pjcoria, 2024-09-03:
Thank you for your answer. Yes I adapted the example you cited. The scpi command has worked fine for me with the V in it for decimation factors I didn't have any problem with, but I fail to see why is that V not letting me use the other deciamtion factors.