RedPitaya/RedPitaya #312: how to use calib tool with the old format

Hermy07, 2024-10-21:
Hi!

is it still possible to use the calib programm in command line with the old format?
As example: 
calib -w 
and then enter just 1 to change the first parameter to 1

b-swat, 2024-10-22:
Hi.
It is impossible to work with individual parameters separately in the old format; the application was poorly made and therefore its functionality was very limited.

```
root@rp-f0c0d2:~# calib
calib version 2.05-384-1c22dfc2a

Usage: calib [OPTION]...

OPTIONS:
 -r    Read calibration values from eeprom (to stdout).
       The -n flag has no effect. The system automatically determines the type of stored data.

 -w    Write calibration values to eeprom (from stdin).
       Possible combination of flags: -wn, -wf, -wfn, -wmn, -wfmn

 -f    Use factory address space.
 -d    Reset calibration values in eeprom from factory zone. WARNING: Saves automatic to a new format

 -i    Reset calibration values in eeprom by default
       Possible combination of flags: -in , -inf.

 -o    Converts the calibration from the user zone to the old calibration format. For ecosystem version 0.98

 -v    Produce verbose output.
 -h    Print this info.
 -x    Print in hex.
 -u    Print stored calibration in unified format.

 -m    Modify specific parameter in universal calibration
 -n    Flag for working with the new calibration storage format.
```

The new application has a key that converts from the new format to the old one.
 -o    Converts the calibration from the user zone to the old calibration format. For ecosystem version 0.98

calib -wm {ID} {VALUE}

List of ID

`#!/bin/bash

UC_DAC_CH1_GAIN=1
UC_DAC_CH1_OFFSET=2

UC_DAC_CH2_GAIN=3
UC_DAC_CH2_OFFSET=4

UC_DAC_CH1_GAIN_X5=5
UC_DAC_CH1_OFFSET_X5=6

UC_DAC_CH2_GAIN_X5=7
UC_DAC_CH2_OFFSET_X5=8

UC_ADC_CH1_GAIN_1_1=9
UC_ADC_CH1_OFFSET_1_1=10

UC_ADC_CH2_GAIN_1_1=11
UC_ADC_CH2_OFFSET_1_1=12

UC_ADC_CH3_GAIN_1_1=13
UC_ADC_CH3_OFFSET_1_1=14

UC_ADC_CH4_GAIN_1_1=15
UC_ADC_CH4_OFFSET_1_1=16

UC_ADC_CH1_GAIN_1_20=17
UC_ADC_CH1_OFFSET_1_20=18

UC_ADC_CH2_GAIN_1_20=19
UC_ADC_CH2_OFFSET_1_20=20

UC_ADC_CH3_GAIN_1_20=21
UC_ADC_CH3_OFFSET_1_20=22

UC_ADC_CH4_GAIN_1_20=23
UC_ADC_CH4_OFFSET_1_20=24

UC_ADC_CH1_GAIN_1_1_AC=25
UC_ADC_CH1_OFFSET_1_1_AC=26

UC_ADC_CH2_GAIN_1_1_AC=27
UC_ADC_CH2_OFFSET_1_1_AC=28

UC_ADC_CH1_GAIN_1_20_AC=29
UC_ADC_CH1_OFFSET_1_20_AC=30

UC_ADC_CH2_GAIN_1_20_AC=31
UC_ADC_CH2_OFFSET_1_20_AC=32

UC_ADC_CH1_AA_1_1=33
UC_ADC_CH1_BB_1_1=34
UC_ADC_CH1_PP_1_1=35
UC_ADC_CH1_KK_1_1=36

UC_ADC_CH2_AA_1_1=37
UC_ADC_CH2_BB_1_1=38
UC_ADC_CH2_PP_1_1=39
UC_ADC_CH2_KK_1_1=40

UC_ADC_CH3_AA_1_1=41
UC_ADC_CH3_BB_1_1=42
UC_ADC_CH3_PP_1_1=43
UC_ADC_CH3_KK_1_1=44

UC_ADC_CH4_AA_1_1=45
UC_ADC_CH4_BB_1_1=46
UC_ADC_CH4_PP_1_1=47
UC_ADC_CH4_KK_1_1=48

UC_ADC_CH1_AA_1_20=49
UC_ADC_CH1_BB_1_20=50
UC_ADC_CH1_PP_1_20=51
UC_ADC_CH1_KK_1_20=52

UC_ADC_CH2_AA_1_20=53
UC_ADC_CH2_BB_1_20=54
UC_ADC_CH2_PP_1_20=55
UC_ADC_CH2_KK_1_20=56

UC_ADC_CH3_AA_1_20=57
UC_ADC_CH3_BB_1_20=58
UC_ADC_CH3_PP_1_20=59
UC_ADC_CH3_KK_1_20=60

UC_ADC_CH4_AA_1_20=61
UC_ADC_CH4_BB_1_20=62
UC_ADC_CH4_PP_1_20=63
UC_ADC_CH4_KK_1_20=64
`

In the old calib app from 1.04 you need to write all the parameters at once
```

function getDefCalibValues(){
# Default calibration parameters set during the process
FE_CH1_FS_G_HI=2684354
FE_CH2_FS_G_HI=2684354
FE_CH1_FS_G_LO=2684354
FE_CH2_FS_G_LO=2684354
FE_CH1_DC_offs=0
FE_CH2_DC_offs=0
BE_CH1_FS=2684354
BE_CH2_FS=2684354
BE_CH1_DC_offs=0
BE_CH2_DC_offs=0
SOME_eeprom_value=-1430532899 #SOME_eeprom_value is some value in eeprom which is not used for anything but after Crt added hv offset calib values this value also appeard.
FE_CH1_DC_offs_HI=0
FE_CH2_DC_offs_HI=0          #FE_CHx_DC_offs_HI  are dc offset parameters for HV jumper settings
}

```

The order of values. And they are very unfriendly ))

To check the current format and current values, you can use:
```
calib -rv
calib -u
```

Hermy07, 2024-10-23:
Thanks for the reply!
I am currently at version 2.00-18.  
So as example if i want to change the gainCalc value of fast_adc_count_1_1 to "2" the input would be:

1. calib -wmn
2. 9 2

is that correct? 
In that case i get the message: [Error:calib_ConvertEEPROM] Invalid data size: 3080 required 20974.

b-swat, 2024-10-23:
and what is the current calibration version in eeprom?
calib -rv

Hermy07, 2024-10-23:
[image]

Would be the output for calib -rv

calib version: 2.00-18-42fd47896

b-swat, 2024-10-23:
not the version of the application, but the format in which the data is written to the eeprom

Hermy07, 2024-10-23:
[image]

Hermy07, 2024-10-23:
[image]

output for calib -u

b-swat, 2024-10-23:
I can't say for sure from memory right now. Try this:
calib -r > calib.txt
Update value in file.
calib -w $(cat calib.txt)

b-swat, 2024-10-23:
The logic is as follows. read all calibration values ​​into a file or variable. change and write all values ​​again. This logic of working with calibration was previously in the old format

b-swat, 2024-10-23:
try this:

CAL=$(calib -r)
echo $CAL | calib -w

Hermy07, 2024-10-23:
unfortunately the same error message

b-swat, 2024-10-24:
hi. i checked on my side. there is indeed a bug. and it is in all versions.

b-swat, 2024-10-25:
Try calib tool from #403
https://downloads.redpitaya.com/downloads/Unify/nightly_builds/ecosystem-2.05-403-9a0244437.zip
You can take the calib application only from the archive. It does not depend on other libraries

Hermy07, 2024-10-29:
thanks! 
i will try it the next days!