RedPitaya/RedPitaya #314: Synchronization issue between IN channels (RP125-14, LN)

KevinNadaud, 2024-12-14:
Hi,

I have some synchronization issue between channels (dark blue CH1, cyan CH2), see the fig below, both signal should be exactly the same (with some magnitude different indeed but8 at least in phase).
[image]

Does I miss some command to "align" input buffers ?
OS version is 2.05-37.
Here is the python code:
`
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

import pyvisa as visa
import matplotlib.pyplot as plt
import numpy as np
import time

amplitude = 0.5
freq = 1e3
rate = 125e6
decFactor = 64
buffSize = 16384

acqDuration = decFactor*buffSize/rate

# to launch scpi server :# systemctl start redpitaya_scpi &
ip = '10.42.0.75'
port = 5000

rm = visa.ResourceManager('@py')

rp = rm.open_resource("TCPIP0::{0}::{1}::SOCKET".format(
    ip, port), read_termination='\r\n')

print(rp.query('*IDN?'))
#RESET acquisition
rp.write('ACQ:RST')
rp.write('ACQ:DATA:FORMAT ASCII')
rp.write('ACQ:DATA:UNITS VOLTS')
rp.write('ACQ:DEC {0}'.format(decFactor))
rp.write('ACQ:TRig:LEV 0')
rp.write('ACQ:SOUR1:GAIN HV')
rp.write('ACQ:SOUR2:GAIN HV')

rp.write('ACQ:START')

# rp.write('ACQ:TRig:DLY 0')
rp.write('ACQ:TRig:DLY 3100')
rp.write('ACQ:TRig AWG_PE')
rp.write('SOUR1:TRIG:INT')
time.sleep(1)

#RESET generator
rp.write('GEN:RST')
#Put settings
rp.write('SOUR1:FREQ:FIX {0:.3f}'.format(freq))
rp.write('SOUR1:VOLT {0}'.format(amplitude))
rp.write('SOUR1:FUNC SINE')
rp.write('SOUR1:BURS:STAT BURST')        # Mode set to BURST
# rp.write('SOUR1:BURS:STAT CONTINUOUS')        # Mode set to BURST
rp.write('SOUR1:BURS:NCYC 3')            # 3 periods in each burst
# Enable output
rp.write('OUTPUT1:STATE ON')
time.sleep(1)

answer = rp.query('ACQ:TRig:STAT?')
while 'WAIT' in answer:
    time.sleep(0.1)
    answer = rp.query('ACQ:TRig:STAT?')

a = np.array([float(val) for val in rp.query('ACQ:SOUR1:DATA?').strip('{}\n\r').split(',')])
plt.plot(a)
a = np.array([float(val) for val in rp.query('ACQ:SOUR2:DATA?').strip('{}\n\r').split(',')])
plt.plot(a)
`

KevinNadaud, 2024-12-15:
Hi downgrade to 2.04, it works well. I dont know how I can help you or if I just close to report ?

b-swat, 2024-12-16:
Hi. Yes indeed, the code you provided does not output exactly what is needed.
The logic of data capture has changed quite a lot in the latest versions of the ecosystem.
I made a script that implements what you need using the latest version of the ecosystem:

`
#!/usr/bin/python3

import sys
import redpitaya_scpi as scpi
import matplotlib.pyplot as plot

rp_s = scpi.scpi(sys.argv[1])

rp_s.tx_txt('ACQ:RST')
rp_s.tx_txt('ACQ:DATA:FORMAT ASCII')
rp_s.tx_txt('ACQ:DATA:UNITS VOLTS')
rp_s.tx_txt('ACQ:DEC 1')
rp_s.tx_txt('ACQ:TRig:DLY 8192')

rp_s.tx_txt('GEN:RST')
rp_s.tx_txt('SOUR1:VOLT {0}'.format(1))
rp_s.tx_txt('SOUR1:FUNC SINE')
rp_s.tx_txt('SOUR1:BURS:STAT BURST') # Mode set to BURST
rp_s.tx_txt('SOUR1:BURS:NCYC 3') # 3 periods in each burst
rp_s.tx_txt('SOUR1:FREQ:FIX {0:.3f}'.format(100000))
rp_s.tx_txt('OUTPUT1:STATE ON')

rp_s.tx_txt('ACQ:START')
rp_s.tx_txt('ACQ:TRIG AWG_PE')

rp_s.tx_txt('SOUR1:TRIG:INT')

while 1:
    rp_s.tx_txt('ACQ:TRIG:STAT?')
    if rp_s.rx_txt() == 'TD':
        break

while 1:
    rp_s.tx_txt('ACQ:TRIG:FILL?')
    if rp_s.rx_txt() == '1':
        break

rp_s.tx_txt('ACQ:SOUR1:DATA:TRig? 8192,POST_TRIG')
buff_string = rp_s.rx_txt()
buff_string = buff_string.strip('{}\n\r').replace("  ", "").split(',')
buff = list(map(float, buff_string))

plot.plot(buff)
plot.ylabel('Voltage')

rp_s.tx_txt('ACQ:SOUR2:DATA:TRig? 8192,POST_TRIG')
buff_string = rp_s.rx_txt()
buff_string = buff_string.strip('{}\n\r').replace("  ", "").split(',')
buff = list(map(float, buff_string))

plot.plot(buff)
plot.ylabel('Voltage')

plot.show()
`

[image]

KevinNadaud, 2024-12-16:
Thanks for the quick answer. 
I will wait the 2.05-37 to be stable to go for this version.

If I understand well, the main difference (appart the use of redpitaya_scpi) is the use of 
'ACQ:SOUR1:DATA:TRig? 8192,POST_TRIG' ?
This command will be mandatory for the future release ?

Regards,

closed 2024-12-16 by KevinNadaud

b-swat, 2025-10-01:
@drsurfer  @KevinNadaud 

Hi. I double-checked the issue.
Yes, the signal on the second channel is shifted in build 37.

[image]

This should be fixed in the latest release.
This is a screenshot from the latest nightly build. It's not much different from the latest release 43.

[image]

The core logic of the ACQ:SOUR<n>:DATA? command has remained unchanged from version to version. It returns the oldest data.

I don't understand why you're using it, since the data isn't valid.
I've illustrated the chain buffer on the diagram.

B - Data is valid and written before the trigger.
C - Data is valid and written after the trigger.
A and D - Data is invalid. This is garbage in the data buffer.

[image]

The ACQ:SOUR<n>:DATA? command returns data in this order: D,A,B,C

There is no need to take data from blocks A and D, as there could be anything there.

KevinNadaud, 2025-10-01:
Hi, I update to the last stable version (37) and using 'ACQ:SOUR1:DATA:TRig? 16384,POST_TRIG', both signals are synchronized

drsurfer, 2025-10-01:
@b-swat If you're asking me, I'm maintaining an application written by others, so I've accepted many design decisions without questioning everything from scratch.

The code assumes that 16,384 samples of valid data can be read after the trigger (a trigger delay of 8,192 samples is used).
I must say that what I find in the buffer is consistent with what is expected, so it's not exactly "garbage."

My final problem is that I need to maintain compatibility with a number of systems that still have older versions of the operating system, some of which are still 1.04-18 and earlier!

Currently, I've included an if statement that chooses which command to use to read the buffer.

`            if (OSversion.Major==2 && OSversion.Build==37) {`
`                _SCPI.SendCommand($"ACQ:SOUR{(int)ch}:DATA:TRig? 16383,POST_TRIG");`
`            }`
`            else {`
`                _SCPI.SendCommand($"ACQ:SOUR{(int)ch}:DATA?");`
`            }`