RedPitaya/RedPitaya #319: IN2 jittering with OS 2.05-37 but not with OS-2.04-35

martinssonh, 2025-03-07:
Hi,

When generating a sine wave on OUT1 and sampling on IN2, the captured signal jitters a lot when using OS 2.05-37 but not with OS 2.04-35. It does not matter if I acquire IN1 or not. Also triggering on `CH1_PE` or `CH2_PE` makes any difference.
The attached gifs show the acquired signals on IN1 and IN2 when generating a sine wave on OU1. IN1 and IN2 are directly connected to OUT1 with equally long coax cables.

Has some SCPI commads changed or is it a bug in OS 2.05-37?

Best,
Hans

STEMlab 125-14 v1.0
Ecosystem: 2.00-35-aff683518
Linux:  2.04

```
import time
import matplotlib.pyplot as plt
import numpy as np
import Instruments.redpitaya_scpi as scpi
import matplotlib.animation as animation

def animate_frame(frameNbr):
    fig.clear()
    plt.plot(plots[:, frameNbr*2], label='Channel 1')
    plt.plot(plots[:, frameNbr*2 + 1], label='Channel 2')
    plt.title(f'OS {os_version}, Capture No {frameNbr}')
    plt.grid()
    plt.legend(loc='center right')
    plt.xlabel('Samples')
    plt.ylabel('Signal (V)')

# Settings
os_version = '2.04-35'
IP = '192.168.1.119'
signal_freq = 10_000
n_samples = 2**14

# Connect to Red Pitaya
dec = 16
rp_ = scpi.scpi(IP)
rp_.check_error()

# Set up acquisition
sample_nbr = np.arange(n_samples)
rp_.tx_txt('ACQ:RST')
rp_.acq_set(dec=dec,
            gain=['HV', 'HV'],
            averaging=True,
            units='VOLTS',
            sample_format='bin')

# Set up and start generation
rp_.tx_txt('GEN:RST')
rp_.sour_set(chan=1, freq=signal_freq)
rp_.tx_txt('OUTPUT:STATE ON')
rp_.tx_txt('SOUR:TRig:INT')

# Initialize window
plt.ion()
fig = plt.figure()
ax = fig.add_subplot(1, 1, 1)
line1, = ax.plot([], np.zeros_like([]), label='Channel 1')
line2, = ax.plot([], np.zeros_like([]), label='Channel 2')
ax.grid()
ax.legend(loc='center right')
ax.set_autoscalex_on(True)
ax.set_autoscaley_on(True)
ax.set_xlabel('Samples')
ax.set_ylabel('Signal (V)')

# Loop over multiple acquisations
n_captures = 20
plots = np.zeros((n_samples, 2*n_captures))
for ii in range(n_captures):
    # Acquire channel 1
    rp_.tx_txt('ACQ:START')
    time.sleep(0.1)
    rp_.tx_txt('ACQ:TRig CH1_PE')
    # Wait until buffer is filled.
    while True:
        rp_.tx_txt('ACQ:TRIG:FILL?')
        if rp_.rx_txt() == '1':
            break
    chan1 = np.array(rp_.acq_data(chan=1, binary=True, convert=True))

    # Acquire channel 2
    rp_.tx_txt('ACQ:START')
    time.sleep(0.1)
    rp_.tx_txt('ACQ:TRig CH2_PE')
    # Wait until buffer is filled.
    while True:
        rp_.tx_txt('ACQ:TRIG:FILL?')
        if rp_.rx_txt() == '1':
            break
    chan2 = np.array(rp_.acq_data(chan=2, binary=True, convert=True))

    plots[:, ii*2] = chan1
    plots[:, ii*2 + 1] = chan2

    line1.set_xdata(sample_nbr)
    line1.set_ydata(chan1)
    line2.set_xdata(sample_nbr)
    line2.set_ydata(chan2)

    ax.set_title(f'OS {os_version}, Capture No {ii}')
    ax.relim()
    fig.canvas.draw()
    fig.canvas.flush_events()
    time.sleep(0.1)

# Create an animation of the images.
# Stop execution with figure open. To quit, close all figures.
fig = plt.figure()
anime = animation.FuncAnimation(
    fig=fig,
    func=animate_frame,
    frames=n_captures,
    repeat_delay=2000,
    interval=200
)
writergif = animation.PillowWriter(fps=1)
anime.save(f'OS_{os_version}.gif', writer=writergif)
plt.show(block=True)
```

[image]
[image]

Lightsaver7, 2025-04-30:
Hello martinssonh,
Thank you for reporting this. We will fix this with the next official OS release.

drsurfer, 2025-09-30:
Hello @martinssonh I had similar problem, acquiring from IN2 a signal generated on OUT1.

I almost went crazy, than discovered that downgrading OS all worked again.
Then I found this closed thread, same issue [https://github.com/RedPitaya/RedPitaya/issues/314](url)
I don't program in python so I cannot suggest a workaround for you.
In SCPI terms, I had to use  ACQ:SOUR2:DATA:TRig? 16383,POST_TRIG  instead of ACQ:SOUR2:DATA? to read the data. They do basically the same thing.

Frankly speaking I don't know why that issue was closed, while the problem is still alive and kicking.