RedPitaya/RedPitaya #320: Possible typo in example & FCLK speed configuration script not working

LinuxlasGNUleaf, 2025-03-14:
Dear RedPitaya developers.

in [this paragraph](https://redpitaya-knowledge-base.readthedocs.io/en/latest/learn_fpga/4_lessons/StopWatch.html#changing-the-fpga-fabric-clock-speed) of the Stopwatch example, it is claimed that the fabric clock FCLK_CLK0 can be set to frequencies in the range of 100 000 to 2 500 000 000. I believe this is a typo, as the maximum frequency defined in the module in VIVADO is 250 MHz, not 2.5 GHz (!), which seems like a more probable max frequency. If this is indeed a typo, I would suggest you fix it to avoid confusion as I did believe for a short while that the clock could indeed be set to 2.5 GHz :)

While trying to change the frequency with the script given in the example, I also noticed that it does not work anymore. The `amba` directory no longer exists in `/sys/devices/soc0/`, sure, but the (probably equivalent?) `axi` directory, specifically the `f8007000.devcfg`subdir is not writeable at all, no matter whether I am root or not. The SD card was set to writeable with the `rw` command too.

The script 
```bash
devcfg=/sys/devices/soc0/axi/f8007000.devcfg
test -d $devcfg/fclk/fclk0 || echo fclk0 > $devcfg/fclk_export
echo 0 > $devcfg/fclk/fclk0/enable
echo 2500000000 > $devcfg/fclk/fclk0/set_rate
echo 1 > $devcfg/fclk/fclk0/enable
```
produces the output

```
root@rp-xxxxxx:~# sudo ./set-fclk.sh 
./set-fclk.sh: 2: cannot create /sys/devices/soc0/axi/f8007000.devcfg/fclk_export: Permission denied
./set-fclk.sh: 3: cannot create /sys/devices/soc0/axi/f8007000.devcfg/fclk/fclk0/enable: Directory nonexistent
./set-fclk.sh: 4: cannot create /sys/devices/soc0/axi/f8007000.devcfg/fclk/fclk0/set_rate: Directory nonexistent
./set-fclk.sh: 5: cannot create /sys/devices/soc0/axi/f8007000.devcfg/fclk/fclk0/enable: Directory nonexistent
```

Do you have any tips for me on this? I'm still very new to the platform and FPGA in general, so please go easy on me :D

edgar-bonet, 2025-03-14:
I believe this issue should be filed against the [redpitaya-knowledgebase][] repository. Specifically the file [StopWatch.rst][].

[redpitaya-knowledgebase]: https://github.com/RedPitaya/redpitaya-knowledgebase
[StopWatch.rst]: https://github.com/RedPitaya/redpitaya-knowledgebase/blob/master/learn_fpga/4_lessons/StopWatch.rst?plain=1#L281-L289

LinuxlasGNUleaf, 2025-03-14:
Oh that's a good point, I can do that as well! But the (somewhat) related second issue remains, I believe for that one this repository is better, right?

edgar-bonet, 2025-03-14:
@LinuxlasGNUleaf wrote:
> I believe for that one this repository is better, right?

I believe so. So these are two separate issues.

LinuxlasGNUleaf, 2025-03-14:
You're right, I'm changing the title and original post to reflect this.

LinuxlasGNUleaf, 2025-03-24:
So is there still a way to change the clock frequency with the given script or a different method "on-the-fly", without going into VIVADO and messing with the clock frequencies there?